njh/EtherCard #166: DNS queries sent to zero MAC if DNS IP != GW IP and both are on the local subnet

swarren, 2015-01-03:
See identical report in some code that uses ethercard:
https://github.com/solarkennedy/ether_house/issues/1

This issue is fixed in a pull request to that project:
https://github.com/solarkennedy/ether_house/pull/2

client_firmware/libraries/ethercard/dns.cpp function EtherCard::dnsLookup() does this:

while(clientWaitingGw())
This waits until the GW IP's ARP request has been answered. However, DNS requests are sent to the DNS IP not the GW IP. These are usually either the same IP or the DNS IP is remote so packets are sent to the GW IP, so this works out.

However, if the GW and DNS IPs are both in the local network, but different, then this doesn't necessarily wait for the DNS IP's ARP request to be complete, which can result in DNS requests being sent to a zero (not yet detemined) MAC address. Such packets will not be received by the target machine's IP stack, and hence the DNS request will receive no response.

To solve this, we need to modify:

while(clientWaitingGw())
to:

while(clientWaitingDns())
However, that function doesn't exist yet. It should be trivial to implement though.

solarkennedy, 2015-01-03:
I'll try to reproduce+confirm on my branch and report back.

solarkennedy, 2015-01-04:
I can reproduce this bug and confirmed the patch on my branch fixes it. I'll rebase the code on your repo and PR.

PR #167 (2015-01-04): fix ethercard issue with GW IP != DNS IP, but both are LAN IPs
This is a git-am of @swarren 's patch.

Fixes #166.

Pretty sure I've done this right.
closed 2015-01-04 by jcw