njh/EtherCard #169: webClient issue using variables in ether.browseUrl

Ion3, 2015-01-11:
Hi,
I hope somebody can help. This does work:

`ether.browseUrl(PSTR("/input/"), "post.json?node=2&csv=10&apikey=ABC", website, my_callback);`

```c++
    DataCombination = DataType + "10" + ApiKey;
    int DataCombination_len = DataCombination.length() + 1; 
    char DataChar[DataCombination_len];
    DataCombination.toCharArray(DataChar, DataCombination_len);

   ether.browseUrl(PSTR("/input/"), DataChar, website, my_callback);
```

Everything is global defined before `void setup ()`:

```c++
 String DataType = ("post.json?node=2&csv=");
 String ApiKey = ("&apikey=b4e41faa9e51c0be1cb5374b91392646");
 String DataCombination = String();

 char DataChar;
```

The data is used for upload to www.emoncms.org and the following link should be published to the Server:

http://www.emoncms.org/input/post.json?node=2&csv=10&apikey=ABC

In the first example it works.

Can anybody see the reason why it is not working with the String data?

Thanks a lot!

vinaychaddha, 2015-08-09:
dnsLookup as well as browseUrl need information like website and other stoed in PROGMEM and does not work for variables from ram.

I know that in arduino, Serial.println(string) need different handling is string is stored in FLASH memory, so it seems that library has been written to handle FLASH memory information in place of RAM memory variables.

Any solution?

solarkennedy, 2017-04-07:
See #268

echza, 2017-09-20:
Any workaround for this situation?

njh, 2018-08-14:
Closing as a duplicate of #268

closed 2018-08-14 by njh