njh/EtherCard #175: tcpReply does not include data only the header in the reply

ccfiel, 2015-02-05:
I have modified the code of the twitter to access my api. I have successfully post the request but the reply i got is only the header the data is not include. Here is the curl output
```
<= Recv header, 17 bytes (0x11)
0000: HTTP/1.1 200 OK
<= Recv header, 23 bytes (0x17)
0000: vary: Accept-Encoding
<= Recv header, 50 bytes (0x32)
0000: access-control-allow-origin: http://meteor.local
<= Recv header, 35 bytes (0x23)
0000: access-control-allow-methods: PUT
<= Recv header, 44 bytes (0x2c)
0000: access-control-allow-headers: Content-Type
<= Recv header, 25 bytes (0x19)
0000: content-type: text/html
<= Recv header, 37 bytes (0x25)
0000: date: Thu, 05 Feb 2015 07:55:19 GMT
<= Recv header, 24 bytes (0x18)
0000: connection: keep-alive
<= Recv header, 28 bytes (0x1c)
0000: transfer-encoding: chunked
<= Recv header, 2 bytes (0x2)
0000: 
<= Recv data, 32 bytes (0x20)
0000: 15
0004: {.
0008: "status": "OFF".}
001b: 0
001e: 
```
As you can see the last part the the data with 32 bytes. In the tcpReply i got this
```

[Twitter Client]
IP:  192.168.2.124
GW:  192.168.2.1
DNS: 192.168.2.1
SRV: 192.168.2.125
Sending data...
Got a response!
HTTP/1.1 200 OK
vary: Accept-Encoding
access-control-allow-origin: http://meteor.local
access-control-allow-methods: PUT
access-control-allow-headers: Content-Type
content-type: text/html
date: Thu, 05 Feb 2015 07:56:20 GMT
connection: keep-alive
transfer-encoding: chunked

End of response
```
Here is the code 
```
void loop () {
  ether.packetLoop(ether.packetReceive());

  const char* reply = ether.tcpReply(session);
  if (reply != 0) {
    Serial.println("Got a response!");
    Serial.println(reply);
    Serial.println("End of response");

  }
}
```
This is the same as the twitter sample.

Any ideas?

lupinchen, 2015-03-17:
I think chunked encoding / HTTP1.1 is the problem here. Isn't it possible to just communicate
using HTTP1.0?

jcw, 2017-02-12:
Yes, that's HTTP/1.1. EtherCard only supports HTTP/1.0.

closed 2017-02-12 by jcw