njh/EtherCard #176: Rethink class and function names

alsternerd, 2015-02-06:
I wonder why every library other than Ethernet.h and UIPEthernet.h always have to use other names and functions than the standard ones. It would make debugging much simpler if they ALL use the same definitions and input, for example if you want to setup a static ip with UIPEthernet.h or Ethernet.h you just use:
`Ethernet.begin(mac, ip, dns, subnet, ...)` and everything being static BYTES.

While with EtherCard.h one has to use `ether.begin(some octet shit, ip, i don't fucking get what comes next)`

I thinkt hat every Ethernet library has to use the same names and definitions for doing the same stuff.
1. That makes debugging easier.
2. You don't have to review your own code everytime you change the librara just because another chip is used.
3. ???
4. Profit.

alsternerd, 2015-02-06:
Oh I forgot to mention that senseless `ether.begin(700, mac, SS-PINwhatthefuck)` The SS pin for ethernetcards is Pin 10 by default, there's no reason why that would be changed. ( Source: http://arduino.cc/en/Reference/Ethernet)
Also the IP is set with `staticSetup`, but why? There's no need to as it'seasier if it's set with ether.begin or better Ethernet.begin() as there are no changes to be made then.

njh, 2015-02-06:
I think it would help your cause if you edited out the swear words.

jcw, 2015-02-06:
Your frustration shows. Feel free to fork and submit improvements.

alsternerd, 2015-02-06:
It's really frustrating as I have to rewrite other libraries with this one, or yes, fork this one and open yet another project which is badly maintained, as this one seems to be maintained very good. So I thought it would be a good idea to have librarys that can just be changed and the functions and functionality remain the same. (Well, maybe it's better I move over to the arduino forums for that issue.)

sierramike, 2015-03-20:
alsternerd, I don't agree with your comment about the CS pin. SPI is designed so you can connect more than one SPI peripheral to the Arduino and then use the CS pin to choose which one to talk to.
And I'm precisely getting angry against libraries which don't permit changing this pin number in the begin method.

Setting the pin in the h file as a #define is a wrong solution, as from one circuit to another I may not use the same pin, depending of other constraints, especially on an Arduino Uno (Atmega 328) which does not have plethore of pins.

alsternerd, 2015-03-20:
@sierramike What you said is to not change the pin but change it. So the pin then may be defined in the sketch, like it's done with the OneWire-lib.

jcw, 2017-02-12:
EtherCard predates the other Ethernet libraries, so that should explain the differences (and it's not my work, BTW). Closing this issue. Feel free to submit a pull request with improvements.

closed 2017-02-12 by jcw