njh/EtherCard #177: NTP problem (regression?)

lupinchen, 2015-02-10:
I use code (see below) that I found somewhere on the internet to send out an 
NTP request from an arduino with an ENC28J60 ethernet shield. I do not get
a reply. If, however, I use an old version of the library (from July 13, 2014 "f113f28")
I get a reply and everything works fine (this is the latest version that seems to work).

#include <EtherCard.h>      
     
// EtherCard
byte Ethernet::buffer[700];
 
void setup () {
  byte myip[]  = { 10,10,1,3 };
  byte gwip[]  = { 10,10,1,1 };
  byte mymac[] = { 0x74,0xF9,0x69,0xBD,0xDD,0xCA };
  
  Serial.begin(9600);
  Serial.println("\n[NTP test]");
 
  if (ether.begin(sizeof Ethernet::buffer, mymac, 4) == 0) {
    Serial.println( "Failed to access Ethernet controller");
  }
  
  ether.staticSetup(myip, gwip);
  ether.printIp("Allocated IP: ", ether.myip);
  ether.printIp("Gateway IP  : ", ether.gwip); 
}
 
void loop () {
  ether.packetLoop(ether.packetReceive());
 
  Serial.println("start ntp request...");
  if (getNtpTime() == 0) {
    Serial.println("no reply");
  }
  
  delay(1500);
}
 
uint8_t ntpServer[4] = { 130, 88, 212, 143 };
uint8_t ntpMyPort = 123;
unsigned long getNtpTime() {  
  unsigned long timeFromNTP;  
  const unsigned long seventy_years = 2208988800UL;     
 
  ether.ntpRequest(ntpServer, ntpMyPort);
  word beginWait = (word) millis();  
  while((word) millis() - beginWait < 1500UL) {      
    word length = ether.packetReceive();      
    ether.packetLoop(length);      
    if(length > 0 && ether.ntpProcessAnswer(&timeFromNTP, ntpMyPort)) {
      Serial.print("Time from NTP: ");
      Serial.println(timeFromNTP);
      return timeFromNTP - seventy_years;
    }
  }
  return 0;
}

njh, 2015-02-10:
Have you tried using DHCP instead of a static IP? I have had trouble with the changes to the static IP code.

lupinchen, 2015-02-11:
Thanks. If I change to dhcp it also works. There seems to be some problem with static IPs.

lupinchen, 2015-03-20:
Ok. I see the errors of my ways. I thought the code worked before; now it doesn't work so this must
be a regression. But, of course, the netmask is missing in my staticSetup routine. If one does
``staticSetup(myip, gwip, 0, {0xff, 0xff, 0xff, 0})`` everything works.

closed 2015-03-20 by lupinchen