njh/EtherCard #179: Eth module long time stability (not to hang)

Tehnicni, 2015-02-18:
Halloo for a 1st time

Which Arduino Eth module to use to get best long time stability (not to hang) which is most important for my system?
I'm also worried about lightning strike and because of that I'm thinking to use POEth (12V from Eth Switch through 2 unused pair in UTP cable)

About my system
I'm working on idea to build home automatisation system with 6 arduino (mainly Mega) around the house and uATX PC http://www.pcengines.ch/alix1d.htm with LabView 8.2.
I'm in doubt which communication to use, Eth or RS485. Last one is rock stable I thing.

Tehnicni, 2015-02-21:
I will use Ethernet, which IC is better ENC28J60 or  W5100

chuyrg, 2015-04-16:
I solved this by resetting the module every once in a while, using the reset pin on the enc28j60, just need to send a digital write with a low value, and then a digital write in high, after doing this, you might want to use the begin function again, doing this, i've had the module last for 2 days with no hangs.

closed 2017-02-12 by jcw