njh/EtherCard #183: Padding for small packets?

sierramike, 2015-03-20:
I applyed a mod from [atasoyweb](http://www.atasoyweb.net/Modifying-The-EtherCard-Library-To-Handle-Large-TCP-Packets) because I could not have large enough buffers to handle what I wanted to send, and noticed a case when the last packet contained no tcp payload, but should only carry the FIN flag. In that case, the packet was 54 bytes long (14 for Ethernet header, 20 for IP header, 20 for TCP header). And no packet arrived to the destination, causing the connection to stay open. I used Wireshark to check, and noticed that this last packet didn't arrive.

Then I came to the Ethernet specification where it states that an Ethernet packet should have a minimum size of 64 bytes, otherwise it should be ignored. Smaller packets should be padded with 0 bytes. (Avoid sending dirty bytes from the buffer as it is known to be a security leakage).

That said, I altered the library, by editing the writeBuf method from this:

'''C++
static void writeBuf(uint16_t len, const byte* data) {
    enableChip();
    xferSPI(ENC28J60_WRITE_BUF_MEM);
    while (len--)
        xferSPI(*data++);
    disableChip();
}
'''

to this:

'''C++
static void writeBuf(uint16_t len, const byte* data) {
    // Add padding if packet is less than 64 bytes
    uint16_t padding = (len < 64 ? 64 - len : 0);
    enableChip();
    xferSPI(ENC28J60_WRITE_BUF_MEM);
    while (len--)
        xferSPI(*data++);
    while (padding--)
        xferSPI(0);
    disableChip();
}
'''

And now it works, it should be great to update the library with this mod?

Don't know if I can do that myself

lupinchen, 2015-03-20:
I was under the impression that these settings (done in the initialization routine) guarantee that
the ENC28J60 does automatic padding:
````c++
writeOp(ENC28J60_BIT_FIELD_SET, MACON3,
            MACON3_PADCFG0|MACON3_TXCRCEN|MACON3_FRMLNEN);
````
So why is this not working?

njh, 2015-03-22:
@sierramike I have sent lots of very small UDP packets with EtherCard and did not experience any problems... ?

lupinchen, 2015-04-09:
I am also experiencing problems when sending a stream of packets with `httpServerReply_with_flags`. In some siutations the last packet just doesn't make it. Looking at the packetSend-routine in enc28j60.cpp there seems to be a bug. There is no check whether the transmit operation for the previous packet has already finished. 

Usually this is not a problem because in the standard mode of operation you receive a packet and send one reply packet. The receive operation takes enough time so that the previous packet is transmitted before we start a new transmission. If I use `httpServerReply_with_flags` and I am sending a large packet everything might be ok as the checksum computation and the copying
to the receive buffer takes enough time. However, If I am sending a small packet I just might be
too fast... 

Of course, this is just a theory...

@sierramike could you check whether your problem also disappears (without your mod) if you add a small delay before sending the last packet?

thekakester, 2017-08-29:
It looks like it is padding to 60 bytes, but it is ALSO setting the length part of the ethernet packet to 60.
The little fix that @sierramike did would set the "length" part of the packet to the correct length, and then proceed to pad the rest.  More to come... digging a bit.

closed 2018-07-27 by njh