njh/EtherCard #190: EtherCard is ported to STM32F103 micro-controller

Serasidis, 2015-05-28:
Hello,

I ported the **EtherCard** library as well the examples too.
I did my best to keep the library functional. I tested the most of the examples. I had had to modify many of the examples to make them work with the combination of  **Arduino_STM32 project** + the **Arduino 1.6.3 IDE**.

Moreover, I put the library files into the **src** folder and I created the **keywords.txt** and **library.properties** files. 

The ported library can be found on Github:
https://github.com/Serasidis/STM32duino/tree/master/libraries/Serasidis_EtherCard_STM

I hope you find that ported library useful.

Vassilis Serasidis

solarkennedy, 2015-05-28:
Whoa nice!

What type of modifications were necessary to do this port? (especially the examples? seems that they should work out of the box if the base library worked?) It is hard to see the differences easily on github because it wasn't done on a branch or anything.

Do you think it is possible to get these changes back upstream with proper ifdefs in place to make the library automatically choose the right values for things like, io ports, memory, etc? That would ensure that updates to this library will improve *both* platforms.

Serasidis, 2015-05-29:
Hello Kyle,

examples
===
In all examples was replaced the
```#include <EtherCard.h>``` 

with the
```#include <EtherCard_STM.h>```
```#include <SPI.h>```

=
Replaced the line:
```static void my_result_cb (byte status, word off, word len)```

with
```static void my_result_cb (byte status, uint16_t off, uint16_t len)```

=
In **getStaticIP.ino** I included
```// domain name server (dns) address```
```static byte dnsip[] = { 192,168,1,1 };```

and
``` ether.copyIp(ether.dnsip, dnsip);```

=
In **noipClient.ino** was replaced the
```static byte mymac[] = {0xDD,0xDD,0xDD,0x00,0x00,0x01};```

with
```static byte mymac[] = {0x74,0x69,0x69,0x2D,0x30,0x31};```

I had network problem with the first MAC address.

==========

Serasidis, 2015-05-29:
Libraries
=========

Replaced
```#include "EtherCard.h"```

to
```#include "EtherCard_STM.h"```

=
in **enc28j60.cpp** added
```#include <SPI.h>```

Was replaced
```
void ENC28J60::initSPI () {
   pinMode(SS, OUTPUT);
   digitalWrite(SS, HIGH);
    pinMode(MOSI, OUTPUT);
    pinMode(SCK, OUTPUT);
    pinMode(MISO, INPUT);

    digitalWrite(MOSI, HIGH);
    digitalWrite(MOSI, LOW);
    digitalWrite(SCK, LOW);
    SPCR = bit(SPE) | bit(MSTR); // 8 MHz @ 16
    bitSet(SPSR, SPI2X);
} 
```

to
```
void ENC28J60::initSPI () {
   SPI.begin(); 
   SPI.setBitOrder(MSBFIRST);
}
```

Commented out the
```
static void xferSPI (byte data){
...
}
```

and replaced the 
``` xferSPI(...); ```

to
``` SPI.transfer(...); ```

Commented out any use of
``` SPCR ```

Commented out  **cli();** and **sei();**

=
Generally was replaced the default **CS** pin **D8** to pin **PA8**
```uint8_t csPin = 8```

to
```uint8_t csPin = PA8```

=
in **EtherCard_STM.cpp** (Ethercard.cpp) file commented out the **eeprom** and **ltoa** functions.

Serasidis, 2015-05-29:
I hope I described clearly the changes I have made in to the source code.

The use of a **ifdefs** is very good idea for a cross-platform EtherCard library but it needs extra time for that!

guipoletto, 2016-08-08:
So, i have this modified to run in the ESP8266. 

How would i go about publishing this mod-ported library?

solarkennedy, 2016-08-10:
You could publish your fork to github, but even better you could add the appropriate ifdefs and make a pull request?

guipoletto, 2016-08-10:
Yeah, i'm working on that. I'm just learning how to deal with the github process.. xD

guipoletto, 2016-08-10:
OK, the library in https://github.com/guipoletto/ethercard_ESP will compile and work.

There is a dependency, off course (https://github.com/esp8266/Arduino).

I masked some AVR specific settings with "#ifdef __AVR__", so it should not break anything for the Arduino UNO / whatever users. 

I don't have any Atmega boards i could test on tough. (may borrow one tonight).

For now, everything that is not __AVR__ is considered esp8266, mostly because i could not find the correct #ifdef for the platform anywhere. 

Most of the non-AVR code should be cross-compatible with the STM32 family, EXCEPT for the CS pin definitions. 
For that one to be trully cross-platform, i would have to find out the correct #ifdef used for each architecture.

jcw, 2017-02-12:
A pull request would be very much appreciated - I'll be happy to merge it. Closing this for now.

closed 2017-02-12 by jcw

neoblack2, 2017-09-13:
Hi, please change ethercard stm32 in SPI2