njh/EtherCard #192: arduino due cant use enc28j60?

nanmenxue, 2015-06-02:
the library cant work in due,can i resolve this problem?

solarkennedy, 2015-06-02:
Can you provide more context about how it doesn't work?

Have you tried using the equivalent SPI pins for the Due?
http://www.arduino.cc/en/Reference/SPI (Section on "Connections")

If you do get it working can you submit a PR updating the readme on the pin hookup for a Due?

markoj21, 2015-06-19:
I hope this gives you more info, using the simple testDHCP.ino which works on the mega just fine.

C:\Users\***\Documents\Arduino\libraries\ethercard-master\dhcp.cpp: In static member function 'static bool EtherCard::dhcpSetup(const char*, bool)':
C:\Users\***\Documents\Arduino\libraries\ethercard-master\dhcp.cpp:327:61: error: 'strncpy_P' was not declared in this scope
             strncpy_P(hostname, hname, DHCP_HOSTNAME_MAX_LEN);
                                                             ^
Error compiling.

simongerads, 2015-09-07:
Hi, I have the same Issue. Any ideas?

prasimix, 2015-10-13:
Hi, I'm also interested if any plans/progress exists for adding support for Arduino Due. Thanks for great job.

closed 2017-02-12 by jcw

antonays, 2018-02-08:
Similar Issue, 

EtherCard.cpp:14:24: fatal error: avr/eeprom.h: No such file or directory
 #include <avr/eeprom.h>