njh/EtherCard #193: POST size seems to be limited to something around 220 and 230 bytes

loran, 2015-06-03:
It seems that whatever the size of the buffer is it not possible to send POST request bigger than ~220-230 bytes.
Am I wrong?

I discovered this http://www.atasoyweb.net/Modifying-The-EtherCard-Library-To-Handle-Large-TCP-Packets that solves the issue for long GET posts. I didn't tried but I'm quite sure this should solve also POST issue.
Could this be added to the "wonderful" ethercard library?

lupinchen, 2015-06-03:
There are several different ways to send a POST request with this library. Which one particular is not working? I think that you should be able to send a request with length equal to the buffer size without any problem. How long a post request do you want to send?

Currently, it is not possible to send multipacket requests as a client. The only thing that is working
is sending multipacket replies when acting as a server. The above modification will not work correctly if several concurrent requests occur or if packets are duplicated or if ARP requests overwrite your receive buffer etc.  

As opposed to sending out multipacket replies w

loran, 2015-06-03:
First, thanks for your answer and after reading you I realized that it doesn't make sense to do multipacket as a client.
This is my need and the way I found a workaround which is far from being perfect: An ATMega micro controller is aggregating various data from sensors. Every 12 seconds, I preparing a Stash that is sent through Stash::prepare.
My POST size varies, depending on the amount of data obtained from sensors. It goes from around 100bytes to 720bytes.
What I discovered is that if the length of the POST variables (?&var1=value1&var2=value2...) received by the http server is longer than 220-230bytes the ATMega is rebooting OR mixing some variables (mainly the arrays).
The size of my Ethernet::buffer is 500 (which is the size needed to contain the data sent from the server to the micro controller). I've tried to increase the buffer higher than 500 and that doesn't change anything there is always the ~220-230 threshold above which the micro controller is becoming instable.
What I did is to explode the POST vars and always remain under 220bytes. But this also means that it can take me up to 4 calls to send 720bytes.
The ideal for me would be to send the POST vars at once.

lupinchen, 2015-06-03:
Could you post the code that generates the POST request?

loran, 2015-06-03:
Here it is:
```C
Stash stash;
byte sd = stash.create();

stash.print("id=");  
for(byte i=0; i < 6; i++){
  if(id[i]<16) { stash.print("0"); }
  stash.print(id[i],HEX);
}

stash.print("&sid="); stash.print(session_id);
stash.print("&up="); stash.print((unsigned long) round(millis()/1000), HEX);

// here is a for loop to populate the POST
for (byte i = 0; i < sizeOfVariableA; i++){
  if(variableA[i].used == 1){
    stash.print("&u");
    for (byte j=0; j<6; j++) { stash8Bits(variableA[i].id[j]); } // stash8Bits adds leading 0 if value < 16
        stash8Bits(...); // few other stuff that describe var
        stash.print("="); stash.print(variableA[i].value);
    }
  }
}
stash.save();
Stash::prepare(PSTR("POST /getValues/post.php HTTP/1.1 \r\n"
                      "Host: $F \r\n"
                      "Content-Type: application/x-www-form-urlencoded \r\n"
                      "Content-Length: $D \r\n"
                      "\r\n"
                      "$H \r\n"),website, stash.size(), sd);  

session_id = ether.tcpSend();
if (freeCount <= 3) { 
  // http://blog.cuyahoga.co.uk/2012/05/theres-something-wrong-with-my-stash/
  Stash::initMap(56);
} // if (freeCount <= 3)
actual_status = ENC28J60_STATUS_SENT;
```

lupinchen, 2015-06-04:
I am sorry but I don't see a problem with your code. You could try to generate a minimum non-working example that exhibits the problem.  Then you could post the complete code of this example and one could go from there...

loran, 2015-06-08:
@lupinchen thanks for your support. I'm going to write a full code that explains what I'm experiencing and will publish it here

closed 2018-07-28 by njh