njh/EtherCard #197: udpClientSendOnly error

hamsafar, 2015-07-07:
Hi,
when I compile udpClientSendOnly example this error occur:
Arduino: 1.6.4 (Windows 7), Board: "Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"

Build options changed, rebuilding all

udpListener.ino: In function 'void setup()':
udpListener:55: error: invalid conversion from 'void (*)(word, byte*, const char*, word) {aka void (*)(unsigned int, unsigned char*, const char*, unsigned int)}' to 'UdpServerCallback {aka void (*)(unsigned int, unsigned char*, unsigned int, const char*, unsigned int)}' [-fpermissive]
In file included from udpListener.ino:6:0:
D:\Electronics\Softwares\Compiler\AVR\Arduino-1-6-4\libraries\ethercard_master/EtherCard.h:489:17: error:   initializing argument 1 of 'static void EtherCard::udpServerListenOnPort(UdpServerCallback, uint16_t)' [-fpermissive]
     static void udpServerListenOnPort(UdpServerCallback callback, uint16_t port);
                 ^
udpListener:58: error: invalid conversion from 'void (*)(word, byte*, const char*, word) {aka void (*)(unsigned int, unsigned char*, const char*, unsigned int)}' to 'UdpServerCallback {aka void (*)(unsigned int, unsigned char*, unsigned int, const char*, unsigned int)}' [-fpermissive]
In file included from udpListener.ino:6:0:
D:\Electronics\Softwares\Compiler\AVR\Arduino-1-6-4\libraries\ethercard_master/EtherCard.h:489:17: error:   initializing argument 1 of 'static void EtherCard::udpServerListenOnPort(UdpServerCallback, uint16_t)' [-fpermissive]
     static void udpServerListenOnPort(UdpServerCallback callback, uint16_t port);
                 ^
invalid conversion from 'void (*)(word, byte*, const char*, word) {aka void (*)(unsigned int, unsigned char*, const char*, unsigned int)}' to 'UdpServerCallback {aka void (*)(unsigned int, unsigned char*, unsigned int, const char*, unsigned int)}' [-fpermissive]

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

hamsafar, 2015-07-07:
Problem solved.
I change this line:
void udpSerialPrint(word port, byte ip[4], const char *data, word len) {
to:
void udpSerialPrint(uint16_t port, uint8_t ip[4],uint16_t src_port, const char *data, uint16_t len){

Example has error?

tuxmartin, 2015-07-07:
I wrote example about 2 years ago - it worked good.
Ethercard has since evolved. Maybe example is not the actual.
If your code works, please send pull request.

lupinchen, 2015-07-22:
Pull request #160 introduced a slight change to the API. Maybe this causes this issue.

solarkennedy, 2016-11-08:
@hamsafar can you confirm if this is still an issue?

closed 2017-02-12 by jcw