njh/EtherCard #200: Memory trouble?

mindstormsking, 2015-08-22:
Hi,

I get very interesting errors when using the library and the ether.browseUrl function. I suspect a memory issue, but more than 50% of both program memory as well as dynamic memory is available according to the compiler. `freeRam()` also shows >1kB available, so possibly I make a mistake in my program, but I can't find any clue. Below is a code snippet.

    const char Website[] PROGMEM = {"www.mysite.nl"};
    const char GetStr[] PROGMEM = {"?who=%02d&what=%d"};
    byte Ethernet::buffer[400];
    volatile char who = -1;
    volatile char what = -1;
    
    //Some other functions

    static int freeRam () {
    extern int __heap_start, *__brkval; 
    int v; 
    return (int) &v - (__brkval == 0 ? (int) &__heap_start : (int) __brkval); 
    }
    
    void loop()
    {
    //Some statemachine, eventually:
        ether.packetLoop(ether.packetReceive());
        char arg[18];
        memcpy_P(arg, GetStr, 17);
        if(who == -1 || what == -1)
        {      
            what = 9;
            who = 99;
            sprintf(arg, arg, who, what);
            what = -1;
            who = -1;
        }
        else
        {
          sprintf(arg, arg, who, what);
        }
        Serial.println("\n>>> REQ"); //DEBUG
        arg[14] = '\0';
        Serial.println(arg); //DEBUG
        ether.browseUrl(PSTR("/Folder/File.php"), arg, Website, downloadCallback);
  }

The response I sometimes, but not always, get, is

    /Folder/File.php^Š† was not found on www.mysite.nl.

With everytime different bits behind the url. I also tried with `arg[0]='\0';`, but the result is the same. Where does this go wrong?

lcafaro, 2016-02-08:
Hello,
I'm having the exact same issue.
This works:

`ether.browseUrl(PSTR("/foo/"), "bar", website, my_callback);`

... this doesn't:
```
char message[] = "bar";
ether.browseUrl(PSTR("/foo/"), message, website, my_callback);
```

All I get is some random garbage:
[image]

Any hints?

chaosdesignnz, 2016-02-08:
Your issue is the  use of **Char**

see [https://www.arduino.cc/en/Reference/Char]( https://www.arduino.cc/en/Reference/Char)
also [https://www.arduino.cc/en/Reference/Array](https://www.arduino.cc/en/Reference/Array)

which you can specify a length for Char in the **[ ]** after the variable name, but this would only work with fixed length variables if you try 
`Char message[10] = "bar" `
You will get other memory data for the remaining 7 characters.
You can try using **int** with empty **[ ]** and see what result you get

Hope this Helps

lcafaro, 2016-02-08:
Yeah, even though `char msg[] = "foo";` should be equivalent to `char msg[4] = "foo";`, according to:

 * http://www.gnu.org/software/gnu-c-manual/gnu-c-manual.html#Initializing-Arrays
 * http://www.gnu.org/software/gnu-c-manual/gnu-c-manual.html#Arrays-as-Strings

So I guess something strange is going on in the `www_client_internal_datafill_cb` call where the message (`client_urlbuf_var`) is parsed in a PSTR.

Anyway, setting a fixed length worked for me.

Thanks

closed 2016-08-21 by solarkennedy