njh/EtherCard #210: Does ethercard support IPV6?

PatientCoder, 2016-02-05:
Does ethercard support IPV6?  As far as I know the ENC28J60 is one of the only Arduino boards that supports IPV6.  When I look at a sent UDP packet using Wireshark the packet has an IPV6 header then an IPV4 header, then data.

jcw, 2016-02-05:
I doubt it. The code is quite old and was written with minimal overhead and code complexity in mind, as far as I can tell. Are you sure it's not some gateway setup or other adding the IPv6 info?

hubertokf, 2016-02-06:
NodeMCU works with [NodeMCU](http://nodemcu.com/index_en.html) and so do [LoPy](https://www.kickstarter.com/projects/1795343078/lopy-the-lora-wifi-and-bluetooth-iot-development-p), of course, if you can use wifi..

njh, 2016-02-07:
uIP (now part of Contiki) supports IPv6.

And there is the UIPEthernet for Arduino, which is based on uIP:
https://github.com/ntruchsess/arduino_uip

Sadly that doesn't support IPv6 either.

jcw, 2017-02-12:
No, EtherCard has no support for IPv6.

closed 2017-02-12 by jcw

njh, 2017-02-12:
Last year I wrote a library called EtherSia that implements IPv6 for Arduino using the ENC28J60 chipset:

https://github.com/njh/ethersia

jcw, 2017-02-12:
Hey, great! Might be something to add to the README here, as EtherCard is getting a bit old...

jcw, 2017-02-12:
Aw, never mind - already in there - apologies for missing that.