njh/EtherCard #211: EtherCard does not work on Arduino Due, compilation error

jrrsoftware, 2016-02-08:
Hi,
I have a compilation error on my Arduino Due:
`
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\dhcp.cpp: In static member function 'static bool EtherCard::dhcpSetup(const char*, bool)':
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\dhcp.cpp:327:61: error: 'strncpy_P' was not declared in this scope
             strncpy_P(hostname, hname, DHCP_HOSTNAME_MAX_LEN);
                                                             ^
Error compiling.`

there is no other code in my sketch, just the library include:

```
#include <EtherCard.h>

void setup() {
}
void loop() {
}

```

I'm getting more compile errors, whe trying to add the EtherCard libarary from Arduino IDE (using all 3 header files):

```
#include <enc28j60.h>
#include <EtherCard.h>
#include <net.h>

void setup() {
}

void loop() {
}

```

`C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:41:12: error: 'uint8_t' does not name a type
     static uint8_t buffer[]; //!< Data buffer (shared by recieve and transmit)
            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:42:12: error: 'uint16_t' does not name a type
     static uint16_t bufferSize; //!< Size of data buffer
            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:46:12: error: 'uint8_t' does not name a type
     static uint8_t* tcpOffset () { return buffer + 0x36; } //!< Pointer to the start of TCP payload
            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:59:12: error: 'uint8_t' does not name a type
     static uint8_t initialize (const uint16_t size, const uint8_t* macaddr,
            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:71:29: error: 'uint16_t' has not been declared
     static void packetSend (uint16_t len);
                             ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:77:12: error: 'uint16_t' does not name a type
     static uint16_t packetReceive ();
            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:83:26: error: 'uint8_t' has not been declared
     static void copyout (uint8_t page, const uint8_t* data);
                          ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:83:46: error: 'uint8_t' does not name a type
     static void copyout (uint8_t page, const uint8_t* data);
                                              ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:83:55: error: ISO C++ forbids declaration of 'data' with no type [-fpermissive]
     static void copyout (uint8_t page, const uint8_t* data);
                                                       ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:89:25: error: 'uint8_t' has not been declared
     static void copyin (uint8_t page, uint8_t* data);
                         ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:89:39: error: 'uint8_t' has not been declared
     static void copyin (uint8_t page, uint8_t* data);
                                       ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:96:12: error: 'uint8_t' does not name a type
     static uint8_t peekin (uint8_t page, uint8_t off);
            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:147:12: error: 'uint8_t' does not name a type
     static uint8_t doBIST(uint8_t csPin = 8);
            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:156:12: error: 'uint16_t' does not name a type
     static uint16_t readPacketSlice(char* dest, int16_t maxlength, int16_t packetOffset);
            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:164:12: error: 'uint16_t' does not name a type
     static uint16_t enc_malloc(uint16_t size);
            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:169:12: error: 'uint16_t' does not name a type
     static uint16_t enc_freemem();
            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:177:31: error: 'uint16_t' has not been declared
     static void memcpy_to_enc(uint16_t dest, void* source, int16_t num);
                               ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:177:60: error: 'int16_t' has not been declared
     static void memcpy_to_enc(uint16_t dest, void* source, int16_t num);
                                                            ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:184:45: error: 'uint16_t' has not been declared
     static void memcpy_from_enc(void* dest, uint16_t source, int16_t num);
                                             ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master/enc28j60.h:184:62: error: 'int16_t' has not been declared
     static void memcpy_from_enc(void* dest, uint16_t source, int16_t num);
                                                              ^
Error compiling.
`

What I'm doing wrong?
Thanks for help

Jack

chaosdesignnz, 2016-02-09:
Hi Jack

I had the same issue when i started using the  enc28j60

Your problem is a clash between the two libraries **enc28j60** & **EtherCard**

You can't use both, Pick one to use. Also I think that the  version of the IDE also has a part to play, but can't remember which one to use.

Hope this helps

jrrsoftware, 2016-02-09:
Hi,
I tried different combinations of includes. If just one is used:

```
#include <EtherCard.h>

void setup() {
}

void loop() {
}
```

Another compilation error is shown:

```
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\dhcp.cpp: In static member function 'static bool EtherCard::dhcpSetup(const char*, bool)':
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\dhcp.cpp:327:61: error: 'strncpy_P' was not declared in this scope
             strncpy_P(hostname, hname, DHCP_HOSTNAME_MAX_LEN);
                                                             ^
Error compiling.
```

I don't know where the `strncpy_P` function is declared. Anyway, I commented it in the library (just for test) and got another compilation error:

```
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp: In static member function 'static void ENC28J60::initSPI()':
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:253:5: error: 'SPCR' was not declared in this scope
     SPCR = bit(SPE) | bit(MSTR); // 8 MHz @ 16
     ^
In file included from D:\Program Files (x86)\Arduino\hardware\arduino\sam\cores\arduino/Arduino.h:44:0,
                 from C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:12:
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:253:16: error: 'SPE' was not declared in this scope
     SPCR = bit(SPE) | bit(MSTR); // 8 MHz @ 16
                ^
D:\Program Files (x86)\Arduino\hardware\arduino\sam\cores\arduino/wiring_constants.h:93:25: note: in definition of macro 'bit'
 #define bit(b) (1UL << (b))
                         ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:253:27: error: 'MSTR' was not declared in this scope
     SPCR = bit(SPE) | bit(MSTR); // 8 MHz @ 16
                           ^
D:\Program Files (x86)\Arduino\hardware\arduino\sam\cores\arduino/wiring_constants.h:93:25: note: in definition of macro 'bit'
 #define bit(b) (1UL << (b))
                         ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:254:12: error: 'SPSR' was not declared in this scope
     bitSet(SPSR, SPI2X);
            ^
D:\Program Files (x86)\Arduino\hardware\arduino\sam\cores\arduino/wiring_constants.h:87:30: note: in definition of macro 'bitSet'
 #define bitSet(value, bit) ((value) |= (1UL << (bit)))
                              ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:254:18: error: 'SPI2X' was not declared in this scope
     bitSet(SPSR, SPI2X);
                  ^
D:\Program Files (x86)\Arduino\hardware\arduino\sam\cores\arduino/wiring_constants.h:87:49: note: in definition of macro 'bitSet'
 #define bitSet(value, bit) ((value) |= (1UL << (bit)))
                                                 ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp: In function 'void enableChip()':
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:258:9: error: 'cli' was not declared in this scope
     cli();
         ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp: In function 'void disableChip()':
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:264:9: error: 'sei' was not declared in this scope
     sei();
         ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp: In function 'void xferSPI(byte)':
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:268:5: error: 'SPDR' was not declared in this scope
     SPDR = data;
     ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:269:14: error: 'SPSR' was not declared in this scope
     while (!(SPSR&(1<<SPIF)))
[... 50 lines cut ...]
     if (bitRead(SPCR, SPE) == 0)
                       ^
D:\Program Files (x86)\Arduino\hardware\arduino\sam\cores\arduino/wiring_constants.h:86:43: note: in definition of macro 'bitRead'
 #define bitRead(value, bit) (((value) >> (bit)) & 0x01)
                                           ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp: In static member function 'static uint8_t ENC28J60::doBIST(byte)':
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:662:17: error: 'SPCR' was not declared in this scope
     if (bitRead(SPCR, SPE) == 0)
                 ^
D:\Program Files (x86)\Arduino\hardware\arduino\sam\cores\arduino/wiring_constants.h:86:32: note: in definition of macro 'bitRead'
 #define bitRead(value, bit) (((value) >> (bit)) & 0x01)
                                ^
C:\Users\Jacek\Documents\Arduino\libraries\ethercard-master\enc28j60.cpp:662:23: error: 'SPE' was not declared in this scope
     if (bitRead(SPCR, SPE) == 0)
                       ^
D:\Program Files (x86)\Arduino\hardware\arduino\sam\cores\arduino/wiring_constants.h:86:43: note: in definition of macro 'bitRead'
 #define bitRead(value, bit) (((value) >> (bit)) & 0x01)
                                           ^
Error compiling.

```

Any ideas?
Thanks in advance,
Jack

jrrsoftware, 2016-02-09:
P.S.: I use latest version of Arduino IDE (1.7.8) if that matters.

Daniel-dk, 2016-10-19:
The errors you get  (" error: 'SPCR' was not declared in this scope ")  are because teh Ethercard library uses AVR SPI registers directly  and not the Arduino SPI library, 
the registers and usage of AVR is not compatible with ARM.

You might be able to go trough the "hardware interface " where specific AVR SPI registers are used and replace them with SPI library functions. 
 - a simple example is sending data :
` SPDR = *data++; `
can become 
`SPI.transfer(*data++);`

solarkennedy, 2016-10-20:
I could update the docs to make it explicit that this library only supports AVR-based platforms?

Daniel-dk, 2016-10-20:
That would probably solve some misconcptions.
I might have to take on porting this to be compatible with the arduino Zero at some point - if we take on a specific project that allows me to spend time on it and requires a cheap-ish Ethernet interface

solarkennedy, 2016-10-22:
Ok. I'm ok with closing this one as "not supported" (yet). This library predates the arduino SPI library. I don't think a port would be that hard, but as @Daniel-dk says, you kinda need a reason to go down that rabbit-hole.

closed 2016-10-22 by solarkennedy