njh/EtherCard #213: warning: 'prog_char' is deprecated

SunboX, 2016-02-12:
using Arduino 1.6.7 on OS X 10.11.3

Getting this warning when trying to compile:

```bash
In file included from /Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.cpp:12:0:
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
                                         ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.cpp:157:36: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
 void Stash::prepare (PGM_P fmt, ...) {
                                    ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.cpp: In static member function 'static void Stash::extract(uint16_t, uint16_t, void*)':
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.cpp:226:11: warning: 'prog_char' is deprecated (declared at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:350): prog_char type is deprecated. [-Wdeprecated-declarations]
     PGM_P fmt = (PGM_P) *++segs;
           ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.cpp: In static member function 'static void Stash::cleanup()':
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.cpp:294:11: warning: 'prog_char' is deprecated (declared at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:350): prog_char type is deprecated. [-Wdeprecated-declarations]
     PGM_P fmt = (PGM_P) *++segs;
           ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.cpp: At global scope:
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.cpp:318:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
 void BufferFiller::emit_p(PGM_P fmt, ...) {
                                         ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.cpp: In member function 'void BufferFiller::emit_p(const prog_char*, ...)':
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.cpp:367:19: warning: 'prog_char' is deprecated (declared at /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:350): prog_char type is deprecated. [-Wdeprecated-declarations]
             PGM_P s = va_arg(ap, PGM_P);
                   ^
In file included from /Users/xxx/Documents/Arduino/libraries/EtherCard/dhcp.cpp:17:0:
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
                                         ^
In file included from /Users/xxx/Documents/Arduino/libraries/EtherCard/dns.cpp:7:0:
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
                                         ^
In file included from /Users/xxx/Documents/Arduino/libraries/EtherCard/tcpip.cpp:14:0:
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
                                         ^
In file included from /Users/xxx/Documents/Arduino/libraries/EtherCard/udpserver.cpp:8:0:
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
                                         ^
In file included from /Users/xxx/Documents/Arduino/libraries/EtherCard/webutil.cpp:7:0:
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/Users/xxx/Documents/Arduino/libraries/EtherCard/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
                                         ^

Der Sketch verwendet 7.040 Bytes (22%) des Programmspeicherplatzes. Das Maximum sind 30.720 Bytes.
Globale Variablen verwenden 1.077 Bytes (52%) des dynamischen Speichers, 971 Bytes für lokale Variablen verbleiben. Das Maximum sind 2.048 Bytes.
```

murikoz, 2016-02-17:
I also have that problem after compilation(windows 7). but my macbook(el capitan) works well, using the same arduino IDE and ethercard versions as on my PC. so i just compile that on mac. but this is not a solution. and i am also interested in solving that problem.

solarkennedy, 2016-02-20:
Can you verify you are using the latest version of the library? I'm pretty sure this was fixed in #121.

Pimmetje, 2016-04-27:
I just tool the master branch and the webclient example in Arduino 1.6.8

In file included from C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.cpp:12:0:

C:\Users\Pim\Documents\Arduino\libraries\ethercard/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

     static void prepare (PGM_P fmt, ...);

                                        ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

     void emit_p (PGM_P fmt, ...);

                                ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }

                                         ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.cpp:157:36: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

 void Stash::prepare (PGM_P fmt, ...) {

                                    ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.cpp: In static member function 'static void Stash::extract(uint16_t, uint16_t, void*)':

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.cpp:226:11: warning: 'prog_char' is deprecated (declared at c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:350): prog_char type is deprecated. [-Wdeprecated-declarations]

     PGM_P fmt = (PGM_P) *++segs;

           ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.cpp: In static member function 'static void Stash::cleanup()':

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.cpp:294:11: warning: 'prog_char' is deprecated (declared at c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:350): prog_char type is deprecated. [-Wdeprecated-declarations]

     PGM_P fmt = (PGM_P) *++segs;

           ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.cpp: At global scope:

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.cpp:318:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

 void BufferFiller::emit_p(PGM_P fmt, ...) {

                                         ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.cpp: In member function 'void BufferFiller::emit_p(const prog_char*, ...)':

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.cpp:367:19: warning: 'prog_char' is deprecated (declared at c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:350): prog_char type is deprecated. [-Wdeprecated-declarations]

             PGM_P s = va_arg(ap, PGM_P);

                   ^

In file included from C:\Users\Pim\Documents\Arduino\libraries\ethercard\dhcp.cpp:17:0:

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

     static void prepare (PGM_P fmt, ...);

                                        ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

[... 62 lines cut ...]
                                ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }

                                         ^

In file included from C:\Users\Pim\Documents\Arduino\libraries\ethercard\webutil.cpp:7:0:

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

     static void prepare (PGM_P fmt, ...);

                                        ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

     void emit_p (PGM_P fmt, ...);

                                ^

C:\Users\Pim\Documents\Arduino\libraries\ethercard\EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]

     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }

ogarcia, 2016-06-07:
I have the latest version of library and have same issue in Arch Linux with Arduino 1.6.8.

```
In file included from /home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.cpp:12:0:
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
                                         ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.cpp:157:36: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
 void Stash::prepare (PGM_P fmt, ...) {
                                    ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.cpp: In static member function 'static void Stash::extract(uint16_t, uint16_t, void*)':
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.cpp:226:11: warning: 'prog_char' is deprecated (declared at /usr/share/arduino/hardware/tools/avr/avr/include/avr/pgmspace.h:350): prog_char type is deprecated. [-Wdeprecated-declarations]
     PGM_P fmt = (PGM_P) *++segs;
           ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.cpp: In static member function 'static void Stash::cleanup()':
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.cpp:294:11: warning: 'prog_char' is deprecated (declared at /usr/share/arduino/hardware/tools/avr/avr/include/avr/pgmspace.h:350): prog_char type is deprecated. [-Wdeprecated-declarations]
     PGM_P fmt = (PGM_P) *++segs;
           ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.cpp: At global scope:
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.cpp:318:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
 void BufferFiller::emit_p(PGM_P fmt, ...) {
                                         ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.cpp: In member function 'void BufferFiller::emit_p(const prog_char*, ...)':
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.cpp:367:19: warning: 'prog_char' is deprecated (declared at /usr/share/arduino/hardware/tools/avr/avr/include/avr/pgmspace.h:350): prog_char type is deprecated. [-Wdeprecated-declarations]
             PGM_P s = va_arg(ap, PGM_P);
                   ^
In file included from /home/ogarcia/Arduino/libraries/ethercard-master/dhcp.cpp:17:0:
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
                                         ^
In file included from /home/ogarcia/Arduino/libraries/ethercard-master/dns.cpp:7:0:
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
                                         ^
In file included from /home/ogarcia/Arduino/libraries/ethercard-master/tcpip.cpp:14:0:
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
                                         ^
In file included from /home/ogarcia/Arduino/libraries/ethercard-master/udpserver.cpp:8:0:
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:171:40: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     static void prepare (PGM_P fmt, ...);
                                        ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
[... cut ...]
     static void prepare (PGM_P fmt, ...);
                                        ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:240:32: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_p (PGM_P fmt, ...);
                                ^
/home/ogarcia/Arduino/libraries/ethercard-master/EtherCard.h:252:41: warning: 'prog_char' is deprecated [-Wdeprecated-declarations]
     void emit_raw_p (PGM_P p, uint16_t n) { memcpy_P(ptr, p, n); ptr += n; }
```

RusArchi, 2016-06-18:
I'm still having this issue... did anybody find a solution?

solarkennedy, 2016-06-19:
Long term the solution is to convert `prog_char` to `char .. PROGMEM`
https://www.arduino.cc/en/Reference/PROGMEM

PR #236 (2016-08-21): update deprecated PGM_P types to "const char* PROGMEM"
Update deprecated PGM_P types to remove compiler warnings, as suggested by @solarkennedy on the related issue.

Fixes #213
closed 2016-08-21 by solarkennedy