njh/EtherCard #219: isLinkUp returns always true

HugoTheBoss, 2016-03-24:
Working on Arduino, make a simple test in the loop method

if  (!ether.isLinkUp) {
      Serial.println(F("No Link"));
    }
    else {
      Serial.println(F("Link"));
    }

after removing the network cable, the result is still true!
Is it a bug, or I understand the method incorrectly?

nuno-silva, 2016-03-25:
`ether.isLinkUp` (or `EtherCard.isLinkUp`) seems to be inherited from `ENC28J60` and is a method (or function), not an attribute (or variable):
```cpp
class ENC28J60 {
public:
 // ...
    static bool isLinkUp ();
```
see [enc28j60.h#L65](https://github.com/jcw/ethercard/blob/master/enc28j60.h#L65)

Therefore, @HugoTheBoss, please try the following:

```cpp
if ( !ether.isLinkUp() ) {
    Serial.println(F("No Link"));
}
else {
    Serial.println(F("Link"));
}
```

The reason `ether.isLinkUp` is always evaluated to `true` is that `ether.isLinkUp` gives you the address of the `static bool ENC28J60::isLinkUp()` method, which of course is non-zero, making it evaluate to `true`.

HugoTheBoss, 2016-03-25:
Thank you very much! I gave the answer in my question :-)...
I haven´t called the method, just used it as boolean!
Works perfect!

closed 2016-03-25 by HugoTheBoss