njh/EtherCard #220: UDP Send bug ...

axwfae, 2016-03-30:
https://github.com/jcw/ethercard/issues/97
https://github.com/jcw/ethercard/issues/181

I'm sorry, please forgive my broken English, I was using google translation!

I was doing UDP tests, we found a problem simply is not going to update ethercard sendudp () function, specify the UDP server ip mac address!

Cause is supposed to send UDP server packets, sent into the wrong mac address.

There are three cases:
1. Use the dhcp, mac address the issue may become GWIP or DNSIP mac address, because I was with a second server, unsure who I am pointing!

Use static ip, but only when staticSetup (myip), mac address the issue at this time becomes 00: 00: 00: 00: 00: 00

When using static ip, use staticSetup (myip, gwip, dnsip, netmask), mac address will become GWIP or DNSIP mac address

So my current solution is to use mode 3, the UDP server's ip settings in gwip, dnsip, so that you can on a normal pc received ethercard udp packets sent over it!

A recommendation can be added manually ipLookup function (similar dnslookup) to the user himself through this function to call to update arp hisip and its corresponding mac address in tcp / ip in!

Remarks:
You can view tcpip.cpp the EtherCard :: udpPrepare (), doing if (is_lan (myip, dip)) is judged, because UDP in LAN, it will execute setMACandIPs (destmacaddr, dip);

At this dip is udp server's ip, udp server but not destmacaddr mac address, will cause such an error occurs!

If users have ipLookup (), you can do this in some action before, to manually update the look destmacaddr to correct udp server mac address, such a mistake will not happen!

========= The following is a description of Chinese ========= 下面是中文的说明

抱歉请原谅我的破英文，我是使用 google 翻译的！

我在做 UDP 测试时，发现了一个问题，简单的说就是 ethercard 不会去更新 sendudp() 函数中，指定的 UDP server ip 的 mac 地址！

导致原本要发给 UDP server 的封包，变成发给了错误的 mac 地址。

情况有 3 种：
1. 使用 dhcp , 发出的 mac 地址会变成了 GWIP 或 DNSIP 的 mac 地址， 因为我二服务器是一同个，我不确定是指向谁！

使用 static ip , 但是只用 staticSetup(myip) 时，此时发出的 mac 地址会变成 00:00:00:00:00:00

使用 static ip ，使用 staticSetup(myip, gwip, dnsip, netmask) 时， mac 地址会变成 GWIP 或 DNSIP 的 mac 地址

所以我目前的解决方案是使用 3 的方式，把 UDP server 的 ip 设置在 gwip , dnsip ，这样就可以正常的在 pc 上接收到 ethercard 发送过来的 udp 封包了！

建议可以在 tcp/ip 中加一个手动 ipLookup 函数 (类似 dnslookup) 给使用者自已透过这一个函数，来调用 arp 去更新 hisip 和其对应的 mac 地址！

备注：
你可以查看 tcpip.cpp 中的 EtherCard::udpPrepare() ， 在做 if(is_lan(myip,dip)) 判断时，因为 UDP 是在 LAN ，所以会执行 setMACandIPs(destmacaddr, dip);

此时 dip 是 udp server 的 ip ，但是 destmacaddr 并不是 udp server 的 mac 地址，才会导致这样的错误发生！

若是有 ipLookup() 时，就可以在做这一些动作前，手动去更新一下 destmacaddr 到正确的 udp server 的 mac 地址，就不会发生这样的错误了！

axwfae, 2016-03-30:
UDP SERVER IP : 192.168.123.150  MAC:00:0E:7F:A8:C9:F1
GWIP : 192.168.123.1 MAC: F4:CE:38:A5:CE:34
DM9051 :  192.168.123.170 OR DHCP  MAC:74:69:69:2D:30:31

TEST_CASE_1 : USE DHCP 
`

`

axwfae, 2016-03-30:
TEST_CASE_2:USE  staticSetup(myip)

axwfae, 2016-03-30:
TEST_CASE_3:USE  staticSetup(myip, gwip, dnsip, netmask)

axwfae, 2016-03-30:
TEST_CASE_4:USE staticSetup(myip, udp_serverip,  udp_serverip, netmask)

`
:
:

//#define DM9051_DHCP
#ifndef DM9051_DHCP
static byte fixmyip[] = {192,168,123,170};
static byte fixgwip[] = {192,168,123,1};
static byte fixdnsip[] = {192,168,123,1};
static byte udp_serverip[] =  {192,168,123,150};
static byte fixnetmask[] = {255,255,255,0};
#endif

:
:

#ifdef DM9051_DHCP 
  if (!ether.dhcpSetup())
    Serial.println("Failed to get configuration from DHCP");
  else
    Serial.println("DHCP configuration done");
#else
  ether.staticSetup(fixmyip,udp_serverip,udp_serverip,fixnetmask);
#endif
:
:
`

com port output 

nenet IOT Demo
Ethernet controller initialized
IP Address:	192.168.123.170
Netmask:	255.255.255.0
Gateway:	192.168.123.1
DNS Address :	192.168.123.1
DM9051_packetSend_42_:
FF FF FF FF FF FF 74 69 69 2D 30 31 8 6 0 1 8 0 6 4 0 1 74 69 69 2D 30 31 C0 A8 7B AA 0 0 0 0 0 0 C0 A8 7B 1  
DM9051_packetSend_42_:
FF FF FF FF FF FF 74 69 69 2D 30 31 8 6 0 1 8 0 6 4 0 1 74 69 69 2D 30 31 C0 A8 7B AA 0 0 0 0 0 0 C0 A8 7B 1  
DM9051_packetReceive_60_:
74 69 69 2D 30 31 F4 EC 38 A5 EC 34 8 6 0 1 8 0 6 4 0 2 F4 EC 38 A5 EC 34 C0 A8 7B 1 74 69 69 2D 30 31 C0 A8 7B AA 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  
DM9051_packetReceive_60_:
FF FF FF FF FF FF 0 E 7F A8 9C F1 8 6 0 1 8 0 6 4 0 1 0 E 7F A8 9C F1 C0 A8 7B 96 0 0 0 0 0 0 C0 A8 7B AA 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  
DM9051_packetSend_42_:
0 E 7F A8 9C F1 74 69 69 2D 30 31 8 6 0 1 8 0 6 4 0 2 74 69 69 2D 30 31 C0 A8 7B AA 0 E 7F A8 9C F1 C0 A8 7B 96  
DM9051_packetReceive_60_:
74 69 69 2D 30 31 0 E 7F A8 9C F1 8 0 45 0 0 20 B4 7 0 0 80 11 E 34 C0 A8 7B 96 C0 A8 7B AA 4 0 5 39 0 C 19 A5 31 32 33 34 0 0 0 0 0 0 0 0 0 0 0 0 0 0  
1337
192.168.123.150
1024
1234
4
DM9051_packetSend_54_:
F4 EC 38 A5 EC 34 74 69 69 2D 30 31 8 0 45 0 0 28 0 0 40 0 40 11 C2 33 C0 A8 7B AA C0 A8 7B 96 5 39 4 0 0 14 2C 2D 48 65 6C 6C 6F 20 57 6F 72 6C 64 0  
UDP Sent !!
`

mac address is error  (is GWIP OR DNSIP mac address): 

> DM9051_packetSend_54_:
F4 EC 38 A5 EC 34 74 69 69 2D 30

ManfredBartz, 2016-11-14:
if the destination is on the local subnet the library must make an ARP request for the destination IP.
It doesn't.

One would have to add another state machine to implement this.  Doing this properly with timeouts and error handling would be a considerable effort.

A work-around is to send to the local subnet's broadcast address.

closed 2017-02-12 by jcw