njh/EtherCard #224: Problem with static ip

apachebcn, 2016-04-21:
Very strange problem, all my test of ether.tcpSend() fails. 
Simple do nothing. Listen works perfect, ping, but not tcpSend and not browseurl()
If i try with oficial sample "getStaticIp.ino" not works, if i modify some lines to work as DHCP works perfectly.

If i follow ethercard code, and some lines to debug (Serial.print testings) i found that:
go to -> uint8_t EtherCard::tcpSend () 
go to -> clientTcpReq(&tcp_result_cb, &tcp_datafill_cb, hisport);
And no more...
clientTcpReq return session_id number, everytime incresed by 1

Why exactly same operation not works with static ip
I'm using mask 255.255.255.0, and i try without mask

apachebcn, 2016-04-21:
IT'S VERY STRANGE, INSERTING A DELAY JUST AFTER ether.staticSetup is working
Why?
This is not good idea to think it's a real patch and solved

	const char website[] PROGMEM = "192.168.1.19";
	uint8_t hisip[] = { 192,168,1,19 };
	static byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };
	static byte myip[] = { 192,168,1,11 };        
	static byte gwip[] = { 192,168,1,1 };         
	static byte dnsip[] = { 192,168,1,1 };        
	static byte maskip[] = { 255, 255, 255, 0 };  
	ether.copyIp(ether.hisip, hisip); ether.printIp("SRV: ", ether.hisip);
	Serial.print(F("Connecting..."));
	if (ether.staticSetup(myip, gwip, dnsip, maskip))
	{
delay(1000); //WITH THIS LINE WORKS WITH THIS LINE WORKS WITH THIS LINE WORKS 
		ether.printIp("IP:  ", ether.myip);
		ether.printIp("MASK:  ", ether.netmask);  
		ether.printIp("DNS: ", ether.dnsip);  
Serial.println(F("Waitting Gateway"));
		bool gw_found = 0;
		unsigned long time_wait_gw = millis();
		for(;;)
		{
		    ether.packetLoop(ether.packetReceive());
		    if ((millis() - time_wait_gw) > 5000)
		    {
		        break;
		    }
		    if (ether.clientWaitingGw())
		    {
		        gw_found = 1;
		        break;
		    }
		}
		if (gw_found)
		{
Serial.println("Gateway found");
		    ether.printIp("GW:  ", ether.gwip);  
		}
		else
		{
Serial.println("Gateway ERROR");
		}
	}
	else
	{
Serial.println("ERROR IN STATIC SETUP");
	}
....

apachebcn, 2016-04-21:
It seem this little delay it's enought with 100ms
But what sense?

gigi199596, 2016-07-21:
Hey,

I got a similar problem! I've developped a little sketch to control sockets over local network and it works perfectly with DHCP but when I use static IP, connection cannot be reached...

I did a workaround by adding:
`static byte dnsip[] = {192, 168, 1,1};`
and adding `dnsip` into the `ether.staticSetup()`:
`ether.staticSetup(myip, gwip, dnsip);`

So now, the website load only once after that it freezes and connection is stopped.
Does someone have a clue on what is going on here? :sos: :green_heart: 

here's my public repo of this project: [Arduino internet sockets](https://github.com/gigi199596/arduino-internet-sockets)
thanks guys!

jaketeater, 2016-11-22:
I had a similar problem.  I was able to use browseUrl to access a page on my LAN if I used DHCP, however, when I switched to a static IP, it browseUrl stopped working.  (I could still access .httpServerReply, just not outgoing stuff.)

Once I put in the delay, it started working with static IP's  Thanks @apachebcn !

My guess is that it has something to do with the netmask and gateway, since I can access the little http server, but the arduino cannot make an outgoing connection.

maddogmaycry, 2017-01-04:
```c++
static byte mymac[] = {0x74,0x69,0x69,0x2D,0x30,0x31};
static byte myip[] = { 192,168,0,200 };
static byte gwip[] = { 192,168,0,1 };
ether.staticSetup(myip, gwip);
ether.parseIp(ether.hisip, "192.168.0.120");
delay(100);
```

Patriot-stv, 2017-09-22:
Help me please solve the problem with static ip.

I want to ping the local computer. I use for this purpose: arduino nano, ENC28J60 + EtherCard lib.

`
#include <EtherCard.h>
#define STATIC 1  // set to 1 to disable DHCP (adjust myip/gwip values below)

int outputPin = 6; // LED1 to pin 6
int anotherOutputPin = 7; // LED2 to pin 7

// ethernet interface mac address, must be unique on the LAN
static byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };

const static uint8_t myip[] = {192, 168, 1, 209};     
const static uint8_t netmask[] = {255, 255, 255, 0};   
const static uint8_t gwip[] = {192, 168, 1, 229}; 
const static uint8_t dnsip[] = {192, 168, 1, 1};   

byte Ethernet::buffer[700];
static uint32_t timer;
static uint32_t count;
static uint32_t all;
 
// called when a ping comes in (replies to it are automatic)
static void gotPinged (byte* ptr) {
  ether.printIp(">>> ping from: ", ptr);
}
 
void setup () {
  pinMode(outputPin, OUTPUT);
  pinMode(anotherOutputPin, OUTPUT);
  digitalWrite(anotherOutputPin, HIGH);
  digitalWrite(outputPin, HIGH);
  count=0;
  all=0;
   Serial.begin(57600);
  Serial.println("\n[pings]");
  if (ether.begin(sizeof Ethernet::buffer, mymac,10) == 0)
    Serial.println( "Failed to access Ethernet controller");
  
  #if STATIC
  Serial.println( "Getting static IP.");
  if (!ether.staticSetup(myip, gwip, dnsip, netmask)){
    Serial.println( "could not get a static IP");
  
  }
#else

  Serial.println("Setting up DHCP");
  if (!ether.dhcpSetup()){
    Serial.println( "DHCP failed");

  }
#endif
  
  ether.printIp("My IP: ", ether.myip);
  ether.printIp("Netmask: ", ether.netmask);
  ether.printIp("GW IP: ", ether.gwip);
  ether.printIp("DNS IP: ", ether.dnsip);
  ether.parseIp(ether.hisip, "192.168.1.229");
  ether.printIp("SRV: ", ether.hisip);
// call this to report others pinging us
  ether.registerPingCallback(gotPinged);
  
  timer = -9999999; // start timing out right away
  Serial.println();
}
 
void loop () {
  word len = ether.packetReceive(); // go receive new packets
  word pos = ether.packetLoop(len); // respond to incoming pings

   //report whenever a reply to our outgoing ping comes back
   if (len > 0 && ether.packetLoopIcmpCheckReply(ether.hisip)) {
    Serial.print("  ");
    Serial.print((micros() - timer) * 0.001, 3);
    Serial.println(" ms");
    count=0;
    all=0; 
}
 
   
  // ping a remote server once every few seconds
  if (micros() - timer >= 5000000) 
  {
    if ((ether.packetLoopIcmpCheckReply(ether.hisip))==0){
    count=count+1; } 
    ether.printIp("Pinging: ", ether.hisip);
     Serial.println("Count ");
     Serial.println(count);
      Serial.println("all ");
     Serial.println(all);
     timer = micros();
     ether.clientIcmpRequest(ether.hisip);
  if (count>5) 
    {
        digitalWrite(outputPin, LOW);    // turn the LED off by making the voltage LOW
        delay(20000);  
        digitalWrite(outputPin, HIGH);   // turn the LED on (HIGH is the voltage level)
        count=0;    
        all=all+1;
     }  
     if (all>2)
       digitalWrite(outputPin, LOW);  
 
  }
}
`

This code good work using DHCP and dont work with Static IP.

Using this library, can I use a static IP?