njh/EtherCard #225: Server response problem!!

alienturtle, 2016-04-22:
I try to run "twitter.ino" example. Server response HTTP/1.1 404 Site or Page Not Found. 
I did ping successful arduino server from my computer. What problems did I get?
Sorry for my broken english.

solarkennedy, 2016-04-22:
HTTP/1.1 ? Seems odd, as the arduino is doing 1.0 (but I'm not sure why I did it that way)

All the endpoints still look to be up from my side. Can you confirm that the endpoint is valid from your network connection by running `curl -0 -v http://arduino-tweet.appspot.com/update` ?

alienturtle, 2016-04-23:
I did this example with LAN network. also, I did other example that send request to PHP server and still got the same problem.

solarkennedy, 2016-04-23:
Well. 404 probably indicates that the arduino is not hitting the right server. Can you pastebin your code?

alienturtle, 2016-04-23:
`// Demo using DHCP and DNS to perform a web client request.
// 2011-06-08 <jc@wippler.nl> http://opensource.org/licenses/mit-license.php

#include <EtherCard.h>

// ethernet interface mac address, must be unique on the LAN
static byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };

byte Ethernet::buffer[700];
static uint32_t timer;

const char website[] PROGMEM = "192.168.0.102"; //php server

// called when the client request is complete
static void my_callback (byte status, word off, word len) {
  Serial.println(">>>");
  Ethernet::buffer[off+300] = 0;
  Serial.print((const char*) Ethernet::buffer + off);
  Serial.println("...");
}

void setup () {
  Serial.begin(57600);
  Serial.println(F("\n[webClient]"));

  if (ether.begin(sizeof Ethernet::buffer, mymac, 53) == 0) 
    Serial.println(F("Failed to access Ethernet controller"));
  if (!ether.dhcpSetup())
    Serial.println(F("DHCP failed"));

  ether.printIp("IP:  ", ether.myip);
  ether.printIp("GW:  ", ether.gwip);  
  ether.printIp("DNS: ", ether.dnsip);  

  if (!ether.dnsLookup(website))
    Serial.println("DNS failed");
    
  ether.printIp("SRV: ", ether.hisip);
}

void loop () {
  ether.packetLoop(ether.packetReceive());
  
  if (millis() > timer) {
    timer = millis() + 5000;
    Serial.println();
    Serial.print("<<< REQ ");
    ether.browseUrl(PSTR("/temperature/review_data.php?"), "", website, my_callback);
  }
}
`
I use xampp v3.2.2 to run php server.

solarkennedy, 2016-04-23:
Ok. This is a different issue than what was originally described. (twitter.ino)

The issue here is that you cannot use an IP address for the "website" field because it will get "looked up".

If you want to connect to an ip address directly, you must not do a dns lookup. Here is kinda an example:
https://github.com/jcw/ethercard/blob/63c77c12ba664a41e6ca75d0d1bad8aa900c3745/examples/pings/pings.ino#L33-L36

This is my best guess. The thing is, if you are getting a 404 then maybe your router is resolving that ip (mine does). Are you sure `http://192.168.0.102/temperature/review_data.php?` exists? Use your webserver logs to confirm that you are or are not hitting it and at what url.

alienturtle, 2016-04-24:
Sorry about that! I think this is a similar issue. I can access to server from other computers but arduino is not.
2 example have a response: 
```
IP:  192.168.0.100
GW:  192.168.0.1
DNS: 192.168.0.1
SRV: 192.168.0.1

<<< REQ >>>
HTTP/1.1 404 Site or Page Not Found
...
```
if I don't have "dns lookup", the server response:
```
IP:  192.168.0.100
GW:  192.168.0.1
DNS: 192.168.0.1
SRV: 0.0.0.0

<<< REQ 
```
maybe problem on HTTP method?

solarkennedy, 2016-04-24:
Did you used the `parseip` example I mentioned in the previous comment to set the SRV ip correctly?
https://github.com/jcw/ethercard/blob/63c77c12ba664a41e6ca75d0d1bad8aa900c3745/examples/pings/pings.ino#L34

It look like neither of those cases has the correct ip.

alienturtle, 2016-04-25:
Wow!!! This work !!! 
How can I show how grateful I am for what you did? Thank you very very much!!!

closed 2016-04-26 by solarkennedy