njh/EtherCard #226: ether.browseUrl() does not allow other ports than 80?

zgoda, 2016-05-18:
I'm trying to fetch some data from a service in my LAN therefore the machine has no DNS name and is running on custom port. The issue seems similar to [one reported 5 years ago on JeeLabs forum](http://forum.jeelabs.net/node/733.html).

```
const byte API_HOST[] = {192, 168, 0, 21};
const unsigned int API_PORT = 5000;
ether.copyIp(ether.hisip, API_HOST);
ether.hisport = API_PORT;  // this does not change anything, request always fails
ether.browseUrl(PSTR("/api/items"), "?p=1", PSTR("192.168.0.21:5000"), cb);
```

zgoda, 2016-05-18:
Hold on, I may have another problem here.

closed 2016-05-18 by zgoda

jaketeater, 2016-11-22:
Were you able to figure this out?  I am struggling to get browseUrl to connect to a remote port other than 80.

TheLuminary, 2017-10-14:
I can confirm that using ether.hisport does work, at least for me at this point in time.