njh/EtherCard #227: TCP Client Connection via Telnet

delta360, 2016-05-26:
Hi there,
I'm having a bit of a problem.

I'm trying to get a Telnet TCP connection working with the Ethercard library. 

What I want it to do is to connect to a telnet server 10.0.0.10 on port 8888, at which point extract the data and disconnect, display the data in the serial console, and repeat every 5 seconds. Would something like that be possible, I couldn't find or understand any examples given.

Thanks.

solarkennedy, 2016-05-26:
Can you confirm, you want the arduino/ethercard to be the tcp client in this case? (the 8888 server is something else)

Yea I guess we don't have a particular example of this, but conceptually this is very similar to any of the http client examples (twitter) using the `tcpSend` and `tcpReply` functions. You will need to adjust the `hisport` to 8888 and you can drop all of the http headers and junk.

You could also use a lower level `clientTcpReq` with a callback of `tcp_datafill_cb` where you could make a buffer and have that callback fill up the buffer.

delta360, 2016-05-26:
I kind of got it working, it connects to the server just fine but returns nothing. It should return something similar to this:
http://i.imgur.com/cQZyofP.png

Code:
`#include <EtherCard.h>
byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };

const char host[] PROGMEM = "192.168.1.10";

static byte session;
byte Ethernet::buffer[700];
Stash stash;

void setup () {
  Serial.begin(57600);
  Serial.println("[CPU Monitor]");

  if (ether.begin(sizeof Ethernet::buffer, mymac) == 0) 
    Serial.println(F("[ERROR] Failed to access Ethernet Controller."));
  if (!ether.dhcpSetup())
    Serial.println(F("[ERROR] DHCP Connection Failure."));

  ether.hisport = 8888;
  ether.printIp("[INFO] My IP:  ", ether.myip);
  ether.printIp("[INFO] Gateway:  ", ether.gwip);  
  ether.printIp("[INFO] DNS Server: ", ether.dnsip);  
  ether.dnsLookup(host);

  ether.printIp("[INFO] CPU Host IP: ", ether.hisip);
  Serial.println("\n[INFO] Connecting to CPU Monitor Server...");

  session = ether.tcpSend();
  ether.persistTcpConnection(false);
  
}

void loop () {
  ether.packetLoop(ether.packetReceive());
  const char* reply = ether.tcpReply(session);
  
  if (reply != 0) {
    Serial.println("Load: ");
    Serial.println(reply);
  }
}`

I can see in the debug that it is in fact connecting:
http://i.imgur.com/NLW0CNX.png

solarkennedy, 2016-05-28:
I recommend using tcpdump instead of netcat to get a better view.

I also recommend establishing a new connection in the loop, you can't really expect a single tcp connection to stay forever.

If you are connecting to a static ip then you don't *have* to do a dns lookup if you don't want to, you can setup `hisip` directly.

I agree that you are pretty close. A bit of wireshark and more code refinement. (new tcp sessions on each connection attempt) should do it.

delta360, 2016-05-28:
It's okay, I managed to figure it out now...

closed 2016-05-28 by solarkennedy

M0R4L3S41, 2020-09-06:
Hello, good morning afternoon, does anyone here know how I can make my enc28j60 card connect to a mikrotick through telnet?
If you explain to me, what I want is for my arduino uno together with my enc28j60 to connect and the arduino sends a command to the mikrotick console from time to time through telnet