njh/EtherCard #230: EtherCard library doesn't work with Arduino/MySQL connector library

bimcas, 2016-07-21:
Hello,

I have Arduino UNO and ENC28J60 - Ethernet Controllers (http://store.iteadstudio.com/images/produce/Shield/ENC28j6_Ethernet_Interface_Module/Ethernetmodule1.jpg). I downloaded EtherCard library for the ENC28J60 Ethernet Controllers, compatible with Arduino IDE (http://jeelabs.org/pub/docs/ethercard/). Arduino with Ethernet Controllers is working perfectly (DHCP, DNS services, internet connection and etc.). Then I'm trying to use MySQL connector library (https://launchpad.net/mysql-arduino) and connect to MySQL. My code below:

#include <EtherCard.h>
#include "sha1.h"
#include "mysql.h"
/* Setup for Ethernet Library */
byte mac_addr[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress server_addr(10, 0, 0, 1);
byte Ethernet::buffer[700];

/* Setup for the Connector/Arduino */
Connector my_conn; // The Connector/Arduino reference
char user[] = "my_user";
char password[] = "sssssii";
void setup() {
 ether.begin(sizeof Ethernet::buffer, mac_addr, 8);
 Serial.println("Connecting...");
 while (!Serial);
 delay(1000);
 Serial.println("Connecting...");
 if (my_conn.mysql_connect(server_addr, 3306, user, password)){
 Serial.println("Success!");
 } else {
 Serial.println("Connection failed.");
 }
}
void loop() {
}

This code isn't working and I'm getting error messages:
In file included from C:\Arduino\libraries\mysql_connector/mysql.h:40:0,
                          from C:\Arduino\sketch_MySQL2\sketch_MySQL2.ino:12:
C:\Arduino\libraries\Ethernet\src/Ethernet.h:39:22: error: 'EthernetClass Ethernet' redeclared as different kind of symbol extern EthernetClass Ethernet;
                      ^
In file included from C:\Arduino\sketch_MySQL2\sketch_MySQL2.ino:1:0:

C:\Arduino\libraries\ethercard-master/enc28j60.h:187:18: error: previous declaration of 'typedef class ENC28J60 Ethernet' typedef ENC28J60 Ethernet; //!< Define alias Ethernet for ENC28J60
                  ^
exit status 1
Error compiling.

I tried many combinations of code, also other libraries, but no result. I think EtherCard, MySQL connector (maybe Arduino Ethernet) libraries are not compatible with each other. 

**Do you have any idea, how to fix those problems? Thanks**

mydhitz, 2018-05-15:
try esp8266

nuno-silva, 2018-05-15:
@bimcas  the problem is that the MySQL connector library seems to be using Arduino's Ethernet library:
the ether**card** library, as far as I'm aware, has a different API and can not be used as a drop-in replacement for Arduino's Ethernet library. Hence the errors you're getting.

Give [UIPEthernet](https://github.com/ntruchsess/arduino_uip) (Drop in replacement for stock Arduino Ethernet library) a try. Also take a look at some other [libraries for the ENC28J60](https://github.com/jcw/ethercard/blob/master/README.md#related-work).

closed 2018-07-27 by njh