njh/EtherCard #232: How i can get all packages?

alxmamaev, 2016-07-26:
I create python server, that send arduino some data. But arduno get only first package, where only system info about: protocol type, server type and server time. How i can get all packages. I seen similar questions, but i dont understand answer.
Sorry, my english is not good.

nuno-silva, 2016-07-26:
@alxmamaev could you please provide more details on the problem? For example, parts of the code you are running on your arduino and the Python script that cause the problem.

alxmamaev, 2016-07-27:
This is all [code](https://github.com/alxmamaev/server_for_arduino)
Arduino send data to server, and server gives data to script. Script hadling data and return result. But arduino get only first package, where system information about protocol, server type , server time, data result be in other packages.

nuno-silva, 2016-07-27:
It seems that the callback for the `browseUrl` function is called after the first TCP packet of data is received. This is hard-coded in [`tcpip.cpp`](https://github.com/jcw/ethercard/blob/master/tcpip.cpp#L787) (the `callback` is `client_tcp_result_cb`):

```c++
if (tcp_client_state==3 && len>0)
{   //TCP connection established so read data
    if (client_tcp_result_cb) {
        uint16_t tcpstart = TCP_DATA_START; // TCP_DATA_START is a formula
        if (tcpstart>plen-8)
            tcpstart = plen-8; // dummy but save
        uint16_t save_len = len;
        if (tcpstart+len>plen)
            save_len = plen-tcpstart;
        (*client_tcp_result_cb)((gPB[TCP_DST_PORT_L_P]>>5)&0x7,0,tcpstart,save_len); //Call TCP handler (callback) function

        if(persist_tcp_connection)
        {   //Keep connection alive by sending ACK
            make_tcp_ack_from_any(len,TCP_FLAGS_PUSH_V);
        }
        else
        {   //Close connection
            make_tcp_ack_from_any(len,TCP_FLAGS_PUSH_V|TCP_FLAGS_FIN_V);
            tcp_client_state = 6;
        }
        return 0;
    }
}
```
As you can see in the code above, as soon as the TCP connection is established with the server, the callback is called with the data in that first received packet, regardless of the connection being closed of kept alive. However, it seems that the callback is also fired for the next received TCP packets, which will contain the rest of the data. Is that what's happening?

alxmamaev, 2016-07-27:
Yes, arduino get only first package. 
I dont understand this, because my english is bed :(

> However, it seems that the callback is also fired for the next received TCP packets, which will contain the rest of the data. Is that what's happening?

alxmamaev, 2016-07-28:
Maybe you want to see what arduino show me? I can make screeshot arduino and server logs.

alxmamaev, 2016-07-29:
Sorry, i realized that prolem not in all packages. I wil seek ....

closed 2016-07-29 by alxmamaev