njh/EtherCard #234: Receive buffer does not get cleared

marenz2569, 2016-08-06:
The receive buffer should get cleared after reading the packet and saving it to the buffer.
After a while, when no space is left in the receive buffer, new packets will not get written into this buffer.
**This is a problem which causes inability to receive packets after so time**.
This can also be read in the ENC28J60 datasheet under point 7.2.4.
Implementing a fix would also involve the Silicon Eratta Issue 14 from the [ENC28J60 Silicon Eratta sheet](http://ww1.microchip.com/downloads/en/DeviceDoc/80349c.pdf).

njh, 2016-08-06:
Since you updated this issue, I am not sure what is being reported?
That the frame should be released directly after reading the packet?

The reason why this was done was so that the readPacketSlice() function works - which is so that you can process a packet bigger the buffer on your micro-controller.

This functionally was added in 841010e7f2782c3f35ec5a29237f847a4db75d8f with the comment:

> This function allows to read the payload of a packet in several slices,
> that each fit into the global buffer. For this to work the receive
> function has been changed to not free the memory of the current packet
> within the receive buffer. This memory is freed in the next call to
> packetReceive(), instead

marenz2569, 2016-08-06:
Sorry, I have not seen the *readPacketSlice()* function.
Now it makes sense, that the buffer gets cleared on the next call to *packetReceiv()*.

closed 2016-08-06 by marenz2569