njh/EtherCard #238: How can i restart/reinitialize (programm) my ENC28J60?

mazahakka, 2016-08-22:
How can i restart/reinitialize (programm) my ENC28J60? (without restart arduino)
sometimes my ENC28J60 "hang on" and i wanna restart only ENC28J60 (without restart arduino) in programm
sorry for my language :(

guipoletto, 2016-08-22:
There is a Reset pin in the ENC28j60 board, this pin has a pullup of 10k in most boards. 

One way to do it is to pull this pun low, then release it (or pull it high)

This will get the ENC28j60 to reset, and wait for configuration from the arduino. 

Then all you need to do is re-implement the ethernet.start sequence. Something like:

PinMode(ETH_RS_PIN, OUTPUT); // this lets you pull the pin low.
DigitalWrite(ETH_RS_PIN, LOW);   // this resets the ENC28J60 hardware
Delay(100); // this makes sure the ENC28j60 resets OK.
DigitalWrite(ETH_RS_PIN, HIGH); // this makes for a fast rise pulse;
PinMode(ETH_RS_PIN, INPUT) // this releases the pin,(puts it in high impedance); lets the pullup in the board do its job.

Now we just Ethernet.begin(mac_address_byte_array, Ip_address_byte_array);

I recommend wrapping this into a function, so as to make the reset mostly transparent to the rest of your sketch. 

Off course, you're gonna have to allocate (loose) a pin from your arduino to use it as this reset_control.

Now, how do you, from inside of the sketch, identifies a hardware lockup from the ENC28j60 chip?

mazahakka, 2016-08-24:
hello. ty for answer. i use this library https://github.com/andresarmento/modbus-arduino which basic on jcw/ethercard library (for ENC28J60). it works fine but sometime (random) the ENC28J60 "hang on" (

chaosdesignnz, 2016-08-24:
To identify a lock up, do a ping or dns lookup, if it fails then reset, wrap it in A function and call it to test the connection every loop of the main program

mazahakka, 2016-08-24:
Now i know how i can reset the duino via WDT (of because havent free pins for hardware reset the ENC28J60). s1 can put please an example of how 2 ping "himself" or another pc/device?  ethercard/examples/pings/pings.ino performs this function?

sorry for my language again and again :(

chaosdesignnz, 2016-08-24:
I would use the dns lookup function instead of the ping example.

if (!ether.dnsLookup(website))
    Serial.println("DNS failed");
  ether.printIp("Server: ", ether.hisip);
 
Put the above in a function , when it  tells you its failed, reset the module

mazahakka, 2016-08-26:
ty chaosdesignnz i will try.
Reset pin ENC28j60 no TTL +5V?

guipoletto, 2016-08-26:
What kind of board do you have? If you're using a UNO, or any other 5v based board, you shoud pay attention to a couple things:

1- VCC for the ethernet chip must be 3.3v

2- you should use a level translator chip/shield betheen the SCLK, MISO, MOSI, and CS lines. 
(The return signal "MISO" from 3.3 to 5 should work most times), and you can do things like use resistors in series with the MOSI, SCK and CS lines, altough it really depends on the speed of your signals. 

3- the reset pin is active-low, that line you quoted is not absolutelly necessary, it is just there to make sure the device is released from the reset state quicker, but you don't need to actually force the line high, since there should be a pullup of about 10k in there already.

chaosdesignnz, 2016-08-26:
My ENC28J60 is 5v tolerant, I tried to run it off 3.3v but had no success.

You could use a NPN transistor , between Reset and Gnd, with the base connected to a pin in your Arduino, this way it shouldn't matter if it is 5v or 3.3v

here is the Datasheet for reference [http://ww1.microchip.com/downloads/en/DeviceDoc/39662c.pdf](url) 

On pg 59-60, Describes the Reset function, Which is active low as stated by guipoletto. It also has a software reset by SPI on pg 26 & 30, Which is a single Byte command of "11111111" or "255" or "0xFF".

Now whether this will work in your situation or not i do not know. 

Hope this helps.

mazahakka, 2016-09-22:
trying 2 reset ENC28J60 programm or with pin arduino. cant reinitialize the ENC28J60 after restart...
example function lan_reset (programm). after this won ping again... think it problem in https://github.com/andresarmento/modbus-arduino but its dead project looks like...(

void lan_reset() {
///pinMode(ETH_RS_PIN, OUTPUT); // this lets you pull the pin low.
///digitalWrite(ETH_RS_PIN, LOW); // this resets the ENC28J60 hardware
///delay(200); // this makes sure the ENC28j60 resets OK.
///digitalWrite(ETH_RS_PIN, HIGH); // this makes for a fast rise pulse;

pinMode(0, OUTPUT); // CS ENC28j60
digitalWrite(0, LOW);
SPI.transfer(255); // RESET
digitalWrite(0, HIGH);

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
// The IP address for the shield
byte ip[] = { 192, 168, 0, 77 };
///byte dns[] = { 192, 168, 0, 3 };
///byte gateway[] = { 192, 168, 0, 3 };
//Config Modbus IP
///mb.config(mac, ip, dns, gateway);
mb.config(mac, ip);

// Add LAMP1_COIL register - Use addCoil() for digital outputs
mb.addCoil(CBUS_LED);
mb.addCoil(CBUS_ZD);
mb.addCoil(CBUS_KL1);
mb.addCoil(CBUS_KL2);
mb.addCoil(CBUS_ResAlarm);
mb.addCoil(CBUS_Alarm);
mb.addCoil(CBUS_StatusLAN);

// Add SENSOR_IREG register - Use addIreg() for analog Inputs
mb.addIreg(RBUS_TEMP1);
mb.addIreg(RBUS_TEMP2);

}

SimonKre, 2016-10-25:
Just adding my 2 cents;) 

1.
ENC28J60 modules may be 5v tolerant... but - the chip will get very hot, even with a radiator it will be hot. This is not good for stability.

2.
ENC28J60 modules won't work on 3.3V pin from Arduino board. This pin can supply approx. 80mA and the module draws 160-200mA. It should be powered by 3.3v external power supply. And check if it is not less than 3.3v.  Ams1117 3.3v stabilizing modules are very cheap and works great with 5v source.
3.3v powered (from extern. power supply) ENC28J60 will be cool and happy.

3. 
Even if it is powered by 3.3v source it still needs to communicate with 5v arduino logic. It will work directly, but again, it is not optimal. ENC28J60 pin HIGH or 1 has approx. 3v not 5v, and might be recognized by Arduino as 0 not 1. The way to go is to use 5v - 3.3 level shifters (cheap modules).

I think these are the first steps to do if ENC28J60 hangs suddenly.

IMHO resetting should be implemented as fail-safe, not treated as primary solution.

There is also a bug in the chip itself, but it is covered in the library. In case of lost packets, there is a solution in enc28j60.h file, line 190:

`/** Workaround for Errata 13.
*   The transmission hardware may drop some packets because it thinks a late collision
*   occurred (which should never happen if all cable length etc. are ok). If setting
*   this to 1 these packages will be retried a fixed number of times. Costs about 150bytes
*   of flash.
*/
#define ETHERCARD_RETRY_LATECOLLISIONS 0`

SimonKre, 2016-10-28:
@mazahakka 
Are You sure that the reset pin from ENC28J60 is connected to pin 0 on the Arduino side? Pin 0 and 1 are reserved for Arduino serial transmission. You should probably use different one.

mazahakka, 2016-10-30:
> 2.ENC28J60 modules won't work on 3.3V pin from Arduino board. This pin can supply approx. 80mA and the module draws 160-200mA. It should be powered by 3.3v external power supply. And check if it is not less than 3.3v. Ams1117 3.3v stabilizing modules are very cheap and works great with 5v source.
3.3v powered (from extern. power supply) ENC28J60 will be cool and happy.
powered from extern. power supply

> 3. Even if it is powered by 3.3v source it still needs to communicate with 5v arduino logic. It will work directly, but again, it is not optimal. ENC28J60 pin HIGH or 1 has approx. 3v not 5v, and [...]
uses ttl 5v

> Are You sure that the reset pin from ENC28J60 is connected to pin 0 on the Arduino side? Pin 0 and 1 are reserved for Arduino serial transmission. You should probably use different one.
didnt use serial transmission. my device only thinks that his arduino :)

> There is also a bug in the chip itself, but it is covered in the library. In case of lost packets, there is a solution in enc28j60.h file, line 190:
>
> /** Workaround for Errata 13.
> *   The [...]

very interested information thx. will try it

closed 2018-07-27 by njh