njh/EtherCard #242: Can't get DHCP

juergen66, 2016-08-26:
Hello,

I have an ENC28J60 board hooked up to an Arduino Uno as described in the documentation, except that the CS ping is connected to Arduino pin 10. With this setup, I can not get a DHCP address.

A pared down version of the pings example looks like this:`#include <EtherCard.h>

static byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };
byte Ethernet::buffer[700];

void setup () {
  Serial.begin(9600);
  Serial.println("\nInitializing controller...");
  int res = ether.begin(sizeof Ethernet::buffer, mymac, 10);
  
  Serial.println("\nGetting DHCP info...");
  if (!ether.dhcpSetup())
    Serial.println(F("DHCP failed"));

  Serial.println("Got DHCP.");
  ether.printIp("IP:  ", ether.myip);
  ether.printIp("GW:  ", ether.gwip);
}

void loop() {
  //Serial.println(".");
}`

The output looks like this:

> Initializing controller...
> 
> Getting DHCP info...
> DHCP failed
> Got DHCP.
> IP:  0.0.0.0
> GW:  0.0.0.0

There is a long pause after "Getting DHCP info..."

If I run example code from the UIPEthernet with the precisely same hardware setup, everything works fine, so I think the hardware is hooked up correctly.

What am I missing?

Thanks,
 --Juergen

shyamjos, 2016-08-26:
i fixed it by adding a static ip entry with mac id of  ethernet module in my router 

then add below line to disable dhcp
#define STATIC 1  // set to 1 to disable DHCP (adjust myip/gwip values below)

SukkoPera, 2016-08-28:
CS pin defaults to 10 on Arduino uno after 4f4acbe50f9dcf0170ce06954041de7e1da00cd1 committed on Aug 23rd, so you don't even have to specify it anymore.

Please see #243 for something that might help.

solarkennedy, 2016-08-28:
This is fixed in master now if you want to try back and use DHCP.

closed 2016-08-28 by solarkennedy

juergen66, 2016-08-29:
Works beautifully. Thank you!

PR #244 (2016-08-28): Fix DHCP request after change the dhcp cookie variable to PROGMEM
Fixes #242 and #243