njh/EtherCard #243: DHCP does not work anymore after recent changes!

SukkoPera, 2016-08-28:
Title says it all. I have traced the issue to commit 279f5bab3a0831d34c7e41b2728649f4f7512d24, which adds a PROGMEM keyword at line 190 of dhcp.cpp.

Removing that PROGMEM seems to bring things back to normal.

This might actually be the same issue that is causing #242.

solarkennedy, 2016-08-28:
@denisvm can you help confirm? @SukkoPera this line right? https://github.com/jcw/ethercard/blob/279f5bab3a0831d34c7e41b2728649f4f7512d24/dhcp.cpp#L190

I can't think of why PROGMEM could cause this to fail. :(

SukkoPera, 2016-08-28:
Yes, that line. PROGMEM could work, but then you must access those bytes with pgm_read_byte() & Co.

denisvm, 2016-08-28:
@solarkennedy I've just tried it on other routers and I've ran into the same issue. Those bytes weren't being set correctly after the `PROGMEM` modifier. I'm going to submit a patch to correct this issue. Sorry about that

solarkennedy, 2016-08-28:
This is fixed now. Please check out master and try again.

SukkoPera, 2016-08-28:
Everything looks fine now, thanks a lot!

closed 2016-08-28 by SukkoPera