njh/EtherCard #245: DNS fails when using ip address instead of domain name in web client example

shyamjos, 2016-08-29:
const char website[] PROGMEM = "192.198.74.111";

[webClient]
IP:  192.168.1.200
GW:  192.168.1.1
DNS: 8.8.4.4
DNS failed
SRV: 0.0.0.0

nuno-silva, 2016-09-01:
@shyamjos you should not try to lookup an IP address: the `dnsLookup` function expects a *domain name*, which is sent as-is to the DNS server.
As far as I know, if you want to use an IP address directly you must do
```c++
ether.parseIp(ether.hisip, "192.198.74.111");
```
instead of using `dnsLookup`. See [examples/ping](https://github.com/jcw/ethercard/blob/master/examples/pings/pings.ino).

@solarkennedy does it make sense for `dnsLookup` to set the client IP (`ether.hisip`) without issuing DNS a request if it's given an IP address instead of a name? If so, I don't mind implementing it.

solarkennedy, 2016-09-01:
No, I would rather it not. However, simply adding in a commented line explaining the alternative would help future users (hopefully) as they look at this example, if you want to implement it that way.

closed 2016-09-13 by shyamjos