njh/EtherCard #249: Error initializing the library

fusion44, 2016-09-24:
Hi

I'm trying to the an ENC28J60 from Sunfounder running with my Arduino Uno but I'm having trouble initializing it. I've tracked the problem down to [enc28j60.cpp#L384](https://github.com/jcw/ethercard/blob/master/enc28j60.cpp#L384). The function readOp will always return 0 and thus will never break out of the loop.

I've reconnected the pins multiple time so I'm sure they are correct and this happens regardless of whether I chose a static or dynamic IP via DHCP.

Thanks!

matus123, 2016-09-25:
Hi,

I had the same problem as you described, but i got it to work by replacing CS pin from 8 to 10.

try to change CS pin from 8 to 10 on your arduino uno and initialize it with
`ether.begin(sizeof Ethernet::buffer, mymac, 10)`.

fusion44, 2016-09-25:
Thanks! You can even use the PIN 8 but you have to set the pin 8 explicitly otherwise it won't work. Looks like the default value is wrong. 

The culprit seems to be the [SS default value in begin](https://github.com/jcw/ethercard/blob/master/EtherCard.h#L295). When replacing the SS with 8 the library works fine without setting the value explicitly.

I actually have no idea why this even compiles as SS doesn't seem to be defined anywhere, but I'm by far not a c++ expert :)

nuno-silva, 2016-09-26:
@fusion44 `SS` is defined "by Arduino" in `pins_arduino.h` (which defines values that may vary from board to board). For instance, for the "standard" (Uno) board, `SS` is:
```c++
static const uint8_t SS   = 10;
```

Other useful constants, such as `MOSI`, `MISO`, `SCK`, `SDA`, `SCL`, `LED_BUILTIN`, `A0`, `A1`, etc, are also defined there. This file is included by the Arduino IDE when you compile a *sketch*. Check it out at [https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/variants/standard](https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/variants/standard).

fusion44, 2016-09-29:
Thanks everyone!

closed 2016-09-29 by fusion44