njh/EtherCard #254: Subdomain

robertiii, 2016-10-21:
Does the library not work with sub domains? I love that it is smaller but I cannot access for instance "somewhere.somewhere.com." the DNS fails. When connecting to a regular www.somehwrre.com its ok but not with sub domains.

solarkennedy, 2016-10-21:
It does work with subdomains. (in some sense, www is a "subdomain")
Can you say the exact hostname for more context?

nuno-silva, 2016-10-21:
@robertiii I can't resolve that domain name on my machine. Perhaps it's a problem with the name and not Arduino?
```shell
$ host freezer.pianoguy.tk
Host freezer.pianoguy.tk not found: 3(NXDOMAIN)
$ host freezer.pianoguy.tk 8.8.8.8
Using domain server:
Name: 8.8.8.8
Address: 8.8.8.8#53
Aliases: 

Host freezer.pianoguy.tk not found: 3(NXDOMAIN)
```

robertiii, 2016-10-22:
I moved it to pianoguy.tk/ubc/freezerlogger

robertiii, 2016-10-22:
I'm beginning to believe it something weird with the hosting

chaosdesignnz, 2016-10-25:
> Trying to connect to freezer.pianoguy.tk

I get a dns error when trying to go to that site in a web browser

solarkennedy, 2016-10-29:
I'm going to close for now. Please re-open if you can confirm that the website works correctly and the library still does not work as advertised.

closed 2016-10-29 by solarkennedy