njh/EtherCard #257: Send and receive data via telnet

terolando, 2016-11-07:
Hello,
I could not find any example of a connection tcp where I connect to the Arduino via telnet and depending on what I send him answer me something.
I always used the W5100 but with the ping function of this library I will finish what I need.
Thank you.

solarkennedy, 2016-11-07:
No, there isn't really an example like that.
This library doesn't really have the primitives for multi-packet connection state tracking that you would need to have a real conversation with it over telnet.

Maybe you could build on similar to udpServer
https://github.com/jcw/ethercard/blob/3af09e38129c43597f4e0febd37e9c838253249f/udpserver.cpp

Combined with some of the persistenece logic from this example:
https://github.com/jcw/ethercard/blob/master/examples/persistence/persistence.ino

You would certainly need `ether.persistTcpConnection(true);`.

jcw, 2017-02-12:
Indeed - it's unlikely that EtherCard will ever get the ability to perform session/state tracking.

closed 2017-02-12 by jcw