njh/EtherCard #258: Thingspeak example is very slow, have many packet loss and 2 fields are blank.

dpiralis, 2016-11-08:
I dont know why there are so many packet loss and call back the "initialize_ethernet()" function every some seconds. If I click on browser https://api.thingspeak.com/update?api_key=CTWBIYTTYNAJB31G&field1="value" every 15sec (thingsspread limit) it will sent the value for 100%. 
This code has  a packet loss of 70+% It should update values every 15 seconds but it update values after some seconds to minutes.
I want to send a data and return true (send ok) or false(retry),  so it can be less unreliable. A loop is running many times so one random packet can be send on random time and maybe never.
Also field4 and filed8 are not working (It is server problem I think).
Server take 2 different IP on random time 52.7.53.111 and 52.200.157.52 still should I use DNS lookup?
This is my public chanel[ my public chanel]( https://thingspeak.com/channels/175723) look the unstable period of updated values.
My [code ](https://gist.github.com/dpiralis/a5fc36207722766caa2fce3e5c1bd7e6)

solarkennedy, 2016-11-08:
Do you mind trying an older version of the library in case this is due to a regression? See https://github.com/jcw/ethercard/issues/246

dpiralis, 2016-11-09:
I did these modifications but nothing changed.. Dns lookup take more than 10 sec sometimes. Also after these modifications as  said on #246 there are too many [warnings ](https://gist.github.com/dpiralis/22da53113121ed913b76fa1936da1741). I want to send only 3 values per day but some times value is not updated. At least is there any return (true or false) if packet received to thinkspeak?

solarkennedy, 2016-11-09:
In order to be really sure, you will have to use something like wireshark or tcpdump to inspect the packets.

dpiralis, 2016-11-09:
Can you send me old versions of libraries?

solarkennedy, 2016-11-10:
This library doesn't have strict releases, but it is git, you can go back in time as much as you want:
https://github.com/jcw/ethercard/commits/master

dpiralis, 2016-11-12:
With DHCP it is working fine.. it upload values every one minute and on every packet i recieve a reply ulnike on static ip

dpiralis, 2016-11-27:
Well, ` ether.dnsLookup(PSTR("www.google.com")` is not working well with `ether.staticSetup(myip, gwip, dnsip, netmask);`

closed 2018-07-28 by njh