njh/EtherCard #259: 400 bad request nodemcu php apache

istabraqmahmood, 2016-12-05:
hi im using nodemcu trying to connect to mysql data base but i have 400 bad request
please help:

`#include <ESP8266WiFi.h>

const char* ssid     = "SEMO-HOUSE";
const char* password = "19911991";
const char* host = "192.168.0.112";
const unsigned int SENSOR_PIN = A0;
String strURL = "";
void setup()
{
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
 int sensor = analogRead(SENSOR_PIN);
  //String sensor = String(sensor);

    WiFiClient client; 
  if (client.connect(host, 80)) {
 
    Serial.println("connected");

strURL = "GET /connect.php?sensor=";
strURL += (int)sensor;
strURL += " HTTP/1.1\r\n";
client.println(strURL);
client.println("Host: localhost\r\n");
client.print("Connection: close\r\n\r\n");

client.println();
  }
    else {
    Serial.println("–> connection failed/n");
  }
  delay(50000);
  
   if (client.available()) {
     char c = client.read();    
     Serial.print(c);
   }

  // if the server's disconnected, stop the client:
  if (!client.connected()) {
     Serial.println();
     Serial.println("disconnecting.");
     client.stop();

     for(;;)     ;  
   }
}
` 

`<?php 
$servername = "localhost"; 
$username = "root";  
$password = "";

//$sensor = $_GET["sensor"]; 
$conn = mysql_connect($servername, $username, $password);
if ($conn) {
    echo "Connected successfully";
} 
else {
	echo "connection failed";
}

$conndb = mysql_select_db('coe', $conn);

echo  "<br>";

//$sql_insert ="insert into pulses (sensor) values ('$sensor')";
$sql_insert ="insert into pulses (sensor) values ('".$_GET["sensor"]."')";

if($sql_insert){
	echo "insert successfull";
}
else {
	echo "insert failed";
}

$result = mysql_query($sql_insert);
if($result){
	echo "insert successfull";
}
else {
	echo "insert failed" . mysql_error($result);
}

?>`

solarkennedy, 2016-12-05:
@istabraqmahmood wrong github project?

istabraqmahmood, 2016-12-05:
@solarkennedy  oh sorry this is my the first time post in github

closed 2016-12-05 by solarkennedy