njh/EtherCard #260: [HELP WANTED] Which LEDs in the LAN Chip should blink?

mueller-ma, 2016-12-26:
Hi,

I downloaded and connected the Lan chip (HanRun HR911105A) to my Arduino Nano. When powering on, a blue led on the lan chip lights up for less then a second, than nothing happens. I tried several examples, none gets a connection. At my switch, the link to the arduino is allways down. Is my hardware broken?

solarkennedy, 2016-12-26:
Can you confirm you are *not* powering through the Arduino Nano's pins via USB?
I suspect there is not enough power.

mueller-ma, 2016-12-26:
@solarkennedy I have a USB Charger with 5V and 4A
Should this be enough?

jcw, 2016-12-27:
If this is a power issue, then the charger's maximum current may not be the limiting factor: the on-board 3.3V regulator of the Nano may not be able to supply enough current to the ENC28J60 chip on the Ethernet board (it turns the difference between 5V and 3.3V into heat).

If you have a multimeter, you can check whether the 3.3V level remains stable - another hint is when the voltage regulator on the Nano is getting fairly hot.

mueller-ma, 2016-12-27:
When nothing other is connected to the nano the multimeter shows 3,34V, but when I connect the Lan chip, I have only 2.2V

chaosdesignnz, 2016-12-27:
I would try a separate power supply with nothing else connected, I have an atx power supply with 12, 5 and 3v on spring terminals test with.

jcw, 2016-12-27:
Ok, that's the problem: the Nano cannot supply enough current on its 3.3V feed to the ENC28J60.

You need another **regulated** 3.3V power supply (not less, and definitely not more) to connect to power on the your network board. The current is probably somewhere around 300..400 mA.

I'm closing this issue for now, as it's most likely to be resolved when you get the power supply issue resolved. Feel free to reopen if that's not the case.

closed 2016-12-27 by jcw