njh/EtherCard #262: Tcp connect w/o gateway?

maddogmaycry, 2017-01-04:
Hi There!
Server 10.0.0.1
Client 10.0.0.2
It's is possible to connect together w/o using a GW in ethercard? When i configure static IP (or dynamic) w/o chosing a GW, all going to ass :(

jcw, 2017-02-12:
I'd try setting the gateway to 10.0.0.1 ...

JohnOH, 2017-09-28:
Does the library always send packets to the gateway? Packets destined for the local LAN should be sent as presented not redirected to the gateway.

mp-mba, 2017-11-30:
Hi, 

Is there a chance for this to be fixed in the near future?

BR

chunghee, 2023-05-28:
Even though the client connects to the server in local network that has no gateway, 
we should to specify gateway address to Ether.staticSetup()... So I just set the gateway as the server's gateway.

like:  ether.staticSetup(myip, remoteip, 0, 0);  // instead of gw ip.

It works. But, when I wrote my project this bug made me confused.

Other library, EthernetENC works well without g/w setting.

chunghee, 2023-06-12:
ether.staticSetup(myip, remoteip, 0, 0);  still has some problem.

When the client begins, without network connection, it fails to resolve arp (of course), but after connect cable into the device, it still does not work.