njh/EtherCard #263: Only receiving headers in response

OKNoah, 2017-01-12:
Hi. Apologies, but I don't know much about how C works, or Arduino. I'm a software developer, but mostly in Javascript.

```
String response;
byte Ethernet::buffer[700];
const int BUFFER_SIZE = JSON_OBJECT_SIZE(5);

static void response_handler (byte status, word off, word len) {
  Ethernet::buffer[off + 300] = 0;
  response = String((char*) Ethernet::buffer + off);
  Serial.println(response);
}
```

The output of this is about 300bytes o fheader information, but no json body. It abruptly stops before that. I have tried several REST APIs and always the same result.

This is why I admit I'm ignorant. I have no idea why `off` is added to 300, then part of `Ethernet::buffer` is set to `0`. Why is `Ethernet::buffer` then added to `off`, and how does that generate a `String`? 

I'd really appreciate being pointed in the right direction. Thank you!

OKNoah, 2017-01-13:
So I've learned a bit about this syntax for dealing with pointers. But how can I get the correct size and position of the response? One would think `off` + `len`, but setting that to 0 seems to delete the whole buffer. Also tried `len - 1`, since `off + len` might technically end up one position past then end of the buffer. For that matter I tried `len - 2, 3, 4... 50` mostly without success.

OKNoah, 2017-01-16:
I'm not sure, but this may have to do with the Serial output only being able to print so much data in a certain amount of time.

OKNoah, 2017-01-20:
I have made an issue on Stackoverflow: http://stackoverflow.com/questions/41763658/cant-adjust-buffer-to-fit-data

LuBossCzech, 2017-02-15:
I have the same problem with web response.
In home small LAN all ok, header and body message is received. But at work LAN (100 computers) is received only headers from web server response. Response HTTP status is 200 OK.
This is part of my Arduino program:

```
// Declare session variable
byte session;

// Send procedure = prepare Stash, HTTP 1.0 and send packet:
session = ether.tcpSend(); 

//loop ...{
   ether.packetLoop(ether.packetReceive());

   const char* reply = ether.tcpReply(session);
   if (reply != 0) {		
      Serial.println(F("Webserver reply:"));
      Serial.println(reply);
   }
// }... loop
```

OKNoah, 2017-02-16:
@LuBossCzech Not sure if this helps you, but I realized there are several other libraries that work with this ethernet adapter. You can see a comparison of them here: https://www.tweaking4all.com/hardware/arduino/arduino-enc28j60-ethernet/

If you are still having trouble, maybe there's an easier library. I have not had time to resolve my problem with this library and will probably switch to a WiFi module anyway.

LuBossCzech, 2017-02-17:
@OKNoah
For a long time I used UIPEthernet library, but now connection on free remote web servers stop working. With telnet test on command line all ok, but with ENC28J60 connection 'open' is failed. I testied GET and POST commands. Ethercard library working OK only from small home LAN. From medium LAN is received only headers. Probably some new security packet updates instaled on web servers or firewalls :-(

And now is very usable ESP8266 chip. It's AVR killer!
:-)

njh, 2018-07-28:
I suspect this is related to splitting the response over multiple IP packets and the EtherCard code is only handling the first packet.

Sudean11, 2018-11-19:
have

> Hi. Apologies, but I don't know much about how C works, or Arduino. I'm a software developer, but mostly in Javascript.
>
> ```
> String response;
> byte Ethernet::buffer[700];
> const int BUFFER_SIZE = [...]

have you figured out how you get the response json body?