njh/EtherCard #265: UDP listener dont pass more than 19 byte packets

sziti, 2017-02-11:
Hello,

It seems UDP listener only process incoming UDP packets less than 19 bytes.

I use UdpListener example and when I send UDP packet: "123456789012345678"
dipslay it:
dest_port: 3000
src_port: 36412
src_port: 192.168.1.227
data: 123456789012345678

When I send "1234567890123456789" it displays nothing.

Other UDP listener tools receive longer packets correctly.
I use Arduino Mega 2560 board and generic ENC28J60 eth mdule.
Thanks, Peter

jcw, 2017-02-12:
Is your buffer large enough? ([example](http://git.jeelabs.org/jcw/ethercard/src/master/examples/udpListener/udpListener.ino#L21))

sziti, 2017-02-12:
Yes, I tried with default setting (byte Ethernet::buffer[500];) 
and tried to increse to 1500. Same result...

niclasboy, 2017-06-13:
Hey, i have got the same problem! Do you solved it?

sziti, 2017-06-14:
Hello, no i haven't :-(
I've tried to figure out how incoming UDP packets processed in 
this library but it is too complicated for me...

My current workaround is using an esp8266 to receive UDP brodacast 
via wifi and pass them to Arduino Mega board by serial port. 

But it would better to use wired ethernet to receive packets too.

Jcw, could you help to solve it?