njh/EtherCard #266: Static IP while using DHCP to get Gateway, DNS & NetMask

VoidTyphoon, 2017-02-19:
HI, 

I'm quite new to Arduino and this is my first shot at a real project, however, I'm facing a problem.
I'm making a network scanner, Got all of that sorted out and it works on my home network along with a web interface to use it.

I need to be able to set a static ip address as this is a device I will take with me and won't be on the same network however along with that each network will have (Or might have) a different gateway IP, and DNS IP.

I was thinking on using DHCP to get these values then store them as a variable then reset the ENC chip and start it back up with a static using the variables it just pulled with a static IP address.

Am i just over complicating this and is there an easier way or am i on the right track here?
If so, any pointers how one would go about this?

Cheers, -Jack

solarkennedy, 2017-02-19:
This is pretty unusual. Have you considered the alternative of just using DHCP with static lease?

VoidTyphoon, 2017-02-19:
I had not, Is this achievable with the ENC28J60 and this driver?

solarkennedy, 2017-02-19:
DHCP static leases (reservations) are a setting of the router to assign the same IP to a particular MAC. It is independent of the network driver or chipset.

The instructions are going to be router-specific. Here are some generic ones:
http://lifehacker.com/5822605/how-to-set-up-dhcp-reservations-so-you-never-have-to-check-an-ip-address-again

closed 2017-02-19 by solarkennedy

VoidTyphoon, 2017-02-19:
This is my problem. 
It will be taken to different networks where i have NO control over the DHCP server.

solarkennedy, 2017-02-19:
If you don't control the DHCP server, then how are you going to ensure that the static IP you want to configure is available and is within the subnet that is being given out?

VoidTyphoon, 2017-02-19:
I had not thought of that, thank you, you've just given me another day of head scratching  😢  
I will have to figure out a way of getting the IP address from the device, any ideas❓

solarkennedy, 2017-02-19:
There are lots of code sample in the examples folder with things like this. `ether.myip` is the IP of the device.
https://github.com/jcw/ethercard/blob/9c6f56fa0a3f31564d806fd79c6479b46d8d1417/examples/getDHCPandDNS/getDHCPandDNS.ino#L34

VoidTyphoon, 2017-02-19:
Yeah, I'm aware of that however actually hard coding something into contact another device, Unless I sent a POST to my server with the LAN IP and then retrieved it through my client machine however that would be a very slow way to get the IP.