njh/EtherCard #267: csPin for Arduino MEGA with Ethernet Shield ENC28J60

JohnzBallad, 2017-03-05:
Hello guys. 

I'm trying to power up my Ethernet Shield with my MEGA. 
I was sending HTTP Request just fine with UNO but when I plugged MEGA, It wont load.

I'm going to show you a piece of my standard code,.
just to you guys know, I already change the 3rd parameter to 8, 10 and 53.

> `// This demo does web requests via DHCP and DNS lookup.
// 2011-07-05 <jc@wippler.nl> http://opensource.org/licenses/mit-license.php

#include <EtherCard.h>

#define REQUEST_RATE 5000 // milliseconds

// ethernet interface mac address
static byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };
// remote website name
const char website[] PROGMEM = "google.com";

byte Ethernet::buffer[700];
static long timer;

// called when the client request is complete
static void my_result_cb (byte status, word off, word len) {
  Serial.print("<<< reply ");
  Serial.print(millis() - timer);
  Serial.println(" ms");
  Serial.println((const char*) Ethernet::buffer + off);
}

void setup () {
  Serial.begin(57600);
  Serial.println("\n[getDHCPandDNS]");
  
  if (ether.begin(sizeof Ethernet::buffer, mymac) == 0) 
    Serial.println( "Failed to access Ethernet controller");

  if (!ether.dhcpSetup())
    Serial.println("DHCP failed");
  
  ether.printIp("My IP: ", ether.myip);
  // ether.printIp("Netmask: ", ether.mymask);
  ether.printIp("GW IP: ", ether.gwip);
  ether.printIp("DNS IP: ", ether.dnsip);

  if (!ether.dnsLookup(website))
    Serial.println("DNS failed");
  ether.printIp("Server: ", ether.hisip);
  
  timer = - REQUEST_RATE; // start timing out right away
}

void loop () {
    
  ether.packetLoop(ether.packetReceive());
  
  if (millis() > timer + REQUEST_RATE) {
    timer = millis();
    Serial.println("\n>>> REQ");
    ether.browseUrl(PSTR("/foo/"), "bar", website, my_result_cb);
  }
}`

solarkennedy, 2017-03-06:
I've never seen this shield before. Did it come with any documentation for which pins are connected to what?

JohnzBallad, 2017-03-06:
Me either. Nope. Not a paper at all. I bought this in PARAGUAY. 
I'll try connect it as if it was a modul.

njh, 2018-07-27:
Did you ever work out which pin Slave Select was connected to?

JohnzBallad, 2018-08-01:
It's has been a while since my last attemp to make this 2 work together. I actually bought a W5100, it was very very easy to make it works with Arduino UNO or MEGA. 

I dont know how to answer you cuz I really don't remember.  Would it be cool to close this issue?

njh, 2018-08-01:
I am glad you got something working.

Yes, I will close the issue.

closed 2018-08-01 by njh