njh/EtherCard #268: PROGMEM declared makes it impossible to dynamically assign website

jatayu86, 2017-04-07:
Hi,

The website is currently declared as const char* website[] PROGMEM which is essential for the library to work.My application needs to dynamically change website to implement a fall back server functionality.How can this be done?

nuno-silva, 2017-04-07:
Hello @jatayu86 ,

It does not need to be const. You can use `char*`. Just use `fromRam=true` in the [`ether.dnsLookup`](https://github.com/jcw/ethercard/blob/master/EtherCard.h#L547) function so it knows that the string is in ram and not in PROGMEM. For instance (if you need to dynamically change to some pre-defined addresses):
```c
char website1[] = "github.com";
char website2[] = "google.com";

char* website = website1;
if (!ether.dnsLookup(website, true))
    Serial.println("DNS failed");
ether.printIp("Server: ", ether.hisip);

website = website2;
if (!ether.dnsLookup(website, true))
    Serial.println("DNS failed");
ether.printIp("Server: ", ether.hisip);
```

The above is not tested, but it should work.

jatayu86, 2017-04-07:
I tried with your suggestion, dnslookup succeded,however the browseurl fails. It doesn't seem to be firing the right url.Also entire code has a weird behavior ,in the sense some other character arrays which i am printing in console doesnt show the correct data that i declared.Any suggestions would be helpful :)

Code:
char website1[] = "xxx.com";
char *website=website1;
 if (!ether.dnsLookup(website,true))
    Serial.println("DNS failed");
  ether.printIp("SRV: ", ether.hisip);
}

ether.browseUrl(PSTR("/dat?ins=xxx"), NULL, website, my_callback);

Output:
<<< REQ >>>
HTTP/1.1 302 Found
Date: Fri, 07 Apr 2017 10:20:16 GMT
Server: Apache/2.2.15 (Red Hat)
Location: https://$S/dat?ins=xxx
Content-Length: 301
Connection: close
Content-Type: text/html; charset=iso-8859-1

nuno-silva, 2017-04-07:
It seems that the `ether.browseUrl` functions expects the `hoststr` (website) to be in program memory:

[ether.browseUrl](https://github.com/jcw/ethercard/blob/master/tcpip.cpp#L579-L591) uses [www_client_internal_datafill_cb](https://github.com/jcw/ethercard/blob/master/tcpip.cpp#L539-L567) to make the request, which uses [BufferFiller::emit_p(const char* fmt PROGMEM, ...)](https://github.com/jcw/ethercard/blob/master/EtherCard.cpp#L319), which expects strings to be in program memory. Unfortunately I don't think there's a workaround for that.

GonzoAS, 2020-01-30:
Is there any solution yet???

leamendoza97, 2020-02-01:
> Is there any solution yet???
> I need to change website address dynamically as well
Maybe you know this but i'd declare a variable with PROGMEM and around 100 bytes length, then i'd fill with string that i'd want. it sucks but it would work

jatayu86, 2020-02-01:
I am not sure if you can assign a variable to a progmem and make it work
with this library.What i had done is assign two seperate progmem with
static url's and call them individually inside a conditional statement.Of
course this works only if your requirement demands switiching of
urls(domain names) based on certain conditions.Also not an efficient
solution :).

GonzoAS, 2020-02-04:
SOLVED! a tiny cracked solution but it works.

const char website[] PROGMEM = "";

char hisipSD[] // char array with dinamic ip, i only had to concatenate website and hisipSD, has some warnings but it works! 

ether.browseUrl(PSTR("/iBreader?"), str, strcat(website,hisipSD), my_callback);

GonzoAS, 2020-02-12:
> I am not sure if you can assign a variable to a progmem and make it work with this library.What i had done is assign two seperate progmem with static url's and call them individually inside a [...]

hi any opinion about what i did?