njh/EtherCard #269: Can't send udp packet to local network with no gateway.

bestpika, 2017-04-18:
I can't send udp packet to local network with no gateway.
But can send to broadcast address (255.255.255.255).

```
#define PORT 5000

#include <EtherCard.h>
#include <IPAddress.h>

uint8_t macAddr[] = { 0, 0, 0, 0, 255, 1 };
uint8_t ipAddr[] = { 172, 16, 255, 1 };
uint8_t netMask[] = { 255, 255, 0, 0 };

uint8_t distAddr[] = { 255, 255, 255, 255 }; // I can't send packet to 172.16.255.2.

byte Ethernet::buffer[256];

void setup() {
  ether.begin(sizeof(Ethernet::buffer), macAddr);
  ether.staticSetup(ipAddr, 0, 0, netMask);
  ether.printIp("", ether.myip);
}

void loop() {
  char c[] = { 'h', 'i' };
  ether.sendUdp(c, sizeof(c), PORT, distAddr, PORT);
  delay(1000);
}
```

solarkennedy, 2017-04-18:
This may need a fix similar to #167.

But udp already does this
https://github.com/jcw/ethercard/blob/master/tcpip.cpp#L377

Can you confirm you are using the latest version of the library?

bestpika, 2017-04-19:
I use `git clone https://github.com/jcw/ethercard.git` to clone this repo.
So I think I get the latest version.

solarkennedy, 2017-04-21:
It is possible that we have a /24 assumption somewhere in the code. 

Can you get a packet capture (tcpdump -s 1500) of the thing trying to transmit packets? (I know this is difficult if they are not arriving at 172.16.255.2, but maybe they are being sent out incorrectly)

Tzoreol, 2017-05-20:
Same problem with repo clone today.