njh/EtherCard #273: Timeout settings for dhcpSetup

jatayu86, 2017-04-30:
How can i adjust the timeout settings of ether.dhcpsetup() when initialising the board? Also i get junk characters when printing Serial out a dynamic variable inside loop() just after the board gets timedout(dhcp failed).Any solution/workaround to these,i already tried giving a delay after dhcp setup,also free memory is around 300 bytes.

solarkennedy, 2017-04-30:
Here is the timeout 
https://github.com/jcw/ethercard/blob/9c6f56fa0a3f31564d806fd79c6479b46d8d1417/dhcp.cpp#L93

300 bytes sounds pretty low. Here is a pretty good article on saving ram on an arduino
https://learn.adafruit.com/memories-of-an-arduino/optimizing-sram

jatayu86, 2017-05-01:
Sorry its not 300 bytes,rather its 300kb. Thanks :)

njh, 2018-07-28:
Looks like this was resolved, so closing ticket.

closed 2018-07-28 by njh