njh/EtherCard #275: Simple timeout for "ether.browseUrl"

selamhabibi, 2017-05-11:
I want to install a timeout function for a simple WebClient () so that a request is discarded after 5 seconds and a corresponding message can be print out via serial.println for example. But I'm not sure if this is already supported by the function "ether.browseUrl" or how this could be implemented?

Basically I use the code examples from the [(WebClient )](https://github.com/jcw/ethercard/blob/master/examples/webClient/webClient.ino) :

```
// called when the client request is complete
static void my_callback (byte status, word off, word len) {
  Serial.println(">>>");
  Ethernet::buffer[off+300] = 0;
  Serial.print((const char*) Ethernet::buffer + off);
  Serial.println("...");
}

...

void loop () {
  ether.packetLoop(ether.packetReceive());
  
  if (millis() > timer) {
    timer = millis() + 5000;
    Serial.println();
    Serial.print("<<< REQ ");
    ether.browseUrl(PSTR("/foo/"), "bar", website, my_callback);
  }
}
```

I am grateful for every suggestion!

solarkennedy, 2017-05-11:
You could use a global variable to represent whether it worked or not, and make `my_callback` set that variable to true and make the timeout loop break if that variable is true.

thekakester, 2017-11-08:
However, with that suggestion, you could potentially send out a different browseURL request following that, and then the first response might trickle in.

I came here looking for a similar question.  I'm trying to poll a website every 10 seconds, but every once in a while if a response takes more than 10 seconds, I get a jumble of responses back, causing ethercard to crash.

selamhabibi, 2017-11-09:
I chose the solution from [Roger Clark](http://www.rogerclark.net/aurduino-ethercard-multiple-browser-request-example/). About different statuses you can "simulate" a timeout.

njh, 2018-07-28:
Looks like you found a solution to your problem, so I have closed this issue.

closed 2018-07-28 by njh