njh/EtherCard #276: Need EtherCard::httpPost example.

bestpika, 2017-05-11:
Could you give me a httpPost example?

↓ I tried, but not work.
```
char *contentType = "Content-Type: application/x-www-form-urlencoded";
String postDataStr = "...";
char *postDataChar = postDataStr.c_str();
ether.httpPost(PATH, HOST, contentType, postDataChar, NULL);
```

solarkennedy, 2017-05-11:
What didn't work about it?

As an alternative, you can construct the POST content and tcp send it directly
https://github.com/jcw/ethercard/blob/master/examples/twitter/twitter.ino#L44

bestpika, 2017-05-12:
```c++
#define HOST "www.gscc.net.tw"
ether.dnsLookup(HOST); // fail
```
```c++
const char HOST[] = "www.google.com.tw";
ether.dnsLookup(HOST); // okay
```
I think this is the point...

bestpika, 2017-05-12:
```cpp
#define HOST "www.gscc.net.tw"
ether.dnsLookup(HOST); // fail
```
```cpp
const char HOST[] PROGMEM = "www.google.com.tw";
ether.dnsLookup(HOST); // okay
```
I think this is the point...

solarkennedy, 2017-05-12:
CNAMEs are currently not currently supported (from the previous www.gscc.net.tw comment, see https://github.com/jcw/ethercard/issues/24)

Yes, currently the dnsLookup function requres a const char:
https://github.com/jcw/ethercard/blob/master/dns.cpp#L91

Is this the nature of your issue? Is it actually POST related?

bestpika, 2017-05-12:
@solarkennedy I use stash now, and It can work.
Thanks.

closed 2017-05-12 by bestpika