njh/EtherCard #277: Ethercard post - twitter.ino

3bb, 2017-06-09:
Hello, I try to make a POST request to my webserber using twitter.ino example. My problem is that I want more than one request so I moved the function call to loop but in this case I don't receive any response anymore.

Initial code : 
`#include <EtherCard.h>
#include <stdint.h>
#include <Wire.h>

// OAUTH key from http://arduino-tweet.appspot.com/
#define api   "TEST"

// ethernet interface mac address, must be unique on the LAN
byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };

const char website[] PROGMEM = "MYSITE.COM";
const char path[] PROGMEM = "api/";

static byte session;

byte Ethernet::buffer[700];
Stash stash;

static void sendToTwitter () {
  Serial.println("Sending tweet...");
  byte sd = stash.create();

  //const char tweet[] = "@solarkennedy the test Twitter sketch works!";
  int temp=21;
  stash.print("apiKey=");
  stash.print(api);
  stash.print("&T=");
  stash.print(temp);
  stash.save();
  int stash_size = stash.size();

  // Compose the http POST request, taking the headers below and appending
  // previously created stash in the sd holder.
  Stash::prepare(PSTR("POST http://$F/$F HTTP/1.0" "\r\n"
    "Host: $F" "\r\n"
    "Content-Length: $D" "\r\n"
    "Content-Type: application/x-www-form-urlencoded" "\r\n"
    "\r\n"
    "$H"),
  website,path, website, stash_size, sd);

  // send the packet - this also releases all stash buffers once done
  // Save the session ID so we can watch for it in the main loop.
  session = ether.tcpSend();
}

void setup () {
  Serial.begin(57600);
  Serial.println("\n[Twitter Client]");

  if (ether.begin(sizeof Ethernet::buffer, mymac) == 0) 
    Serial.println(F("Failed to access Ethernet controller"));
  if (!ether.dhcpSetup())
    Serial.println(F("DHCP failed"));

  ether.printIp("IP:  ", ether.myip);
  ether.printIp("GW:  ", ether.gwip);  
  ether.printIp("DNS: ", ether.dnsip);  

  if (!ether.dnsLookup(website))
    Serial.println(F("DNS failed"));

  ether.printIp("SRV: ", ether.hisip);

  sendToTwitter();
}

void loop () {
  ether.packetLoop(ether.packetReceive());

  const char* reply = ether.tcpReply(session);
  if (reply != 0) {
    Serial.println("Got a response!");
    Serial.println(reply);
  }
}`

with this code I got this response, which is ok :
`
[Twitter Client]
IP:  192.168.1.4
GW:  192.168.1.1
DNS: XXX.XXX.XXX.XX
SRV: XXX.XXX.XXX.XXX
Sending tweet...
Got a response!
HTTP/1.1 200 OK
Server: nginx
Content-Type: text/html; charset=UTF-8
Connection: close
Cache-Control: private, must-revalidate
Date: Fri, 09 Jun 2017 20:04:48 GMT
pragma: no-cache
expires: -1

ok
`

When I moved the call function in loop 
`#include <EtherCard.h>
#include <stdint.h>
#include <Wire.h>

// OAUTH key from http://arduino-tweet.appspot.com/
#define api   "apikeytest1"

// ethernet interface mac address, must be unique on the LAN
byte mymac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 };

const char website[] PROGMEM = "agri-tech.ro";
const char path[] PROGMEM = "api/save2/";

static byte session;

byte Ethernet::buffer[700];
Stash stash;

static void sendToTwitter () {
  Serial.println("Sending tweet...");
  byte sd = stash.create();

  //const char tweet[] = "@solarkennedy the test Twitter sketch works!";
  int temp=21;
  stash.print("apiKey=");
  stash.print(api);
  stash.print("&T=");
  stash.print(temp);
  stash.save();
  int stash_size = stash.size();

  // Compose the http POST request, taking the headers below and appending
  // previously created stash in the sd holder.
  Stash::prepare(PSTR("POST http://$F/$F HTTP/1.0" "\r\n"
    "Host: $F" "\r\n"
    "Content-Length: $D" "\r\n"
    "Content-Type: application/x-www-form-urlencoded" "\r\n"
    "\r\n"
    "$H"),
  website,path, website, stash_size, sd);

  // send the packet - this also releases all stash buffers once done
  // Save the session ID so we can watch for it in the main loop.
  session = ether.tcpSend();
}

void setup () {
  Serial.begin(57600);
  Serial.println("\n[Twitter Client]");

  if (ether.begin(sizeof Ethernet::buffer, mymac) == 0) 
    Serial.println(F("Failed to access Ethernet controller"));
  if (!ether.dhcpSetup())
    Serial.println(F("DHCP failed"));

  ether.printIp("IP:  ", ether.myip);
  ether.printIp("GW:  ", ether.gwip);  
  ether.printIp("DNS: ", ether.dnsip);  

  if (!ether.dnsLookup(website))
    Serial.println(F("DNS failed"));

  ether.printIp("SRV: ", ether.hisip);

  
}

void loop () {
  sendToTwitter();

  delay(1000);
  
 
  ether.packetLoop(ether.packetReceive());

  const char* reply = ether.tcpReply(session);
  if (reply != 0) {
    Serial.println("Got a response!");
    Serial.println(reply);
  }
}`

all I got is this :
`[Twitter Client]
IP:  192.168.1.4
GW:  192.168.1.1
DNS: XXX.XXX.XXX.XX
SRV: XXX.XXX.XXX.XXX
Sending tweet...
Sending tweet...
Sending tweet...
`
Any sugestions  ? Thanks

solarkennedy, 2017-06-10:
Have you tried without `delay(1000);`?

3bb, 2017-06-10:
Hello solarkennedy,
after few hours of debuging I found the problem on 22th page of google :))). When function call is in setup function I make one request and after that I check for response in void loop. Server response isn't instant so you have to wait around 3-4-5 seconds. 
When I moved the call in void setup, it runs too fast. I made one request, check if I got a response than make another request even if last response didn't arrive yet. 
Solution is to make other request after you receive the response : 

> if ( responseReceived == 1 )  {
>  makeRequest ();
> responseReceived = 0;
> }
> 
> ether.packetLoop(ether.packetReceive());
> 
> const char* reply = ether.tcpReply(session);
> if (reply != 0) {
> Serial.println("Got a response!");
> Serial.println(reply);
> responseReceived == 1;
> }

Thank you for help and hope this will help other people !

closed 2017-06-10 by 3bb