njh/EtherCard #278: Problem with DHCP and switches

kiwidoc, 2017-06-12:
There appears to be a problem with the ethercard not being able to pull a DHCP address from my router when there is a switch in the way.

solarkennedy, 2017-06-13:
My best guess would be that the wireless bridge is the core issue. Is it in WDS mode?
If it *isn't* using WDS then MAC addresses are going to be mangled and I wouldn't expect dhcp to work correctly.

Windows (and other full-featured dhcp implementations) probably get away with this by skipping the DISCOVER phase if has dhcp'd before. (just a guess)

The only way to be sure is to get a packet capture of failed DHCP request from the arduino using Wireshark, which may be difficult.

kiwidoc, 2017-06-13:
Thanks

Difficult  and beyond my techincal competence!

I noted someone else having very much the same problem on stackexchange 
https://arduino.stackexchange.com/questions/19335/w5100-ethernet-shield-dhcp-fails-with-a-switch
This one did not have the bridge.

AFAICS I cannot use the Ubiquiti Litebeam 5AC23 pair which form the 
bridge as WDS - the distance is over 100m so most devices intended for 
easy use won't work.  I can always use a fixed ip, which should work I 
think, but I was trying to create a system whereby IOT devices could 
register their presence on a DHCP network and manage such as expiration 
of lease or re-assignment of the IP address without failure.  Just 
experimenting really.  I have plenty of spare fixed IPs to use.

I will try fiddling with some of the settings - the ACs can also act as 
routers which may help to solve the problem but I will have to wait for 
the weekend to do that due to time constraints..

Regards

Rob

njh, 2018-07-28:
Sorry, it has been a number of months and I am not sure anyone is going to be able to help solve this.
I think it really needs a Wireshark packet capture to work out what is going on.

Closing this issue, but please re-open if you have more information or someone finds a solution.

closed 2018-07-28 by njh

johseffer, 2021-07-13:
Same problem here. The communication works when connecting directly on the router. Still working when i connect the ethernet to a switch (Fortinet Firewall) with is connected with the router. But when i tried to disconnect the switch from the router, setting static ips for the devices and trying to make an http request to another device connected to the switch the request never get a response. After some investigation with the wireshark we can see that the ethercard make a request to the configured gateway so he can decide where is the request destination.

ARDUINO IP: 192.168.1.100
DESTINY IP: 192.168.1.200
GATEWAY: 192.168.1.1

With devices connected directly in the router:
REQUEST: 192.168.1.100 -> 192.168.1.200

With switch disconnected from the router:
REQUEST: 192.168.1.100 -> 192.168.1.1 -> 192.168.1.200

In my case, my switch has a setting that block requests from lan x lan :( That is the cause of my problem and I can get it with the wireshark dumping logs of the switch.