njh/EtherCard #279: ether.dnsLookUp fails if is I check it multiple times

3bb, 2017-06-28:
Hello,
I have the following problem :
I made an tcp request, I recevied the server response and after that for 1 minute I don't make other requests but in this time ( in loop function ) dnsLookup is called to check if connection is still alive. The problem is that after first request dnslookup freeze for 30s-1m or maybe more.

solarkennedy, 2017-06-29:
Can you paste a snapshot of the code?

Are you running `ether.packetLoop(ether.packetReceive());` in that loop?

3bb, 2017-06-29:
Hello,

```
void setup() {
 	Serial.begin(57600);

  printf_begin();

  initialize_ethernet();
  RTC.setAlarm(ALM2_EVERY_MINUTE,0,0,0,0);
  RTC.read(tm);

}

void loop(){
	
	 if (RTC.alarm(ALARM_2)) {
	 	 while(millis() - startedAt < 10000){

		 	ether.packetLoop(ether.packetReceive());
		 	if (apiResponse == 1) {
		 	 	// do something
		 	 	session = ether.tcpSend();
		 	 	apiResponseReceived = 0;
		 	 } // end if

		 	const char* reply = ether.tcpReply(session);

		    if (reply != 0) {
		       Serial.println(F(" >>>REPLY recieved...."));
		       Serial.println(reply);
		       apiResponseReceived = 1;
		       Serial.println(apiResponseReceived);
		       break;
		    }
	 	}	// end while 
	} // end if RTC.alarm

	if (!ether.dnsLookup(website)) {
	 initialize_ethernet();
	}
}

void initialize_ethernet(void){
  for(int i=0;i<30;i++){ // keep trying until you succeed
    printf_P(PSTR("incerca conexiune internet : "));
    Serial.print(i);

    if (ether.begin(sizeof Ethernet::buffer, mymac, CSPIN) == 0){
      printf_P(PSTR( "Failed to access Ethernet controller"));
      continue;
    }

    if (!ether.dhcpSetup()){
      Serial.println("DHCP failed");
      continue;
    }

    ether.printIp("IP:  ", ether.myip);
    ether.printIp("GW:  ", ether.gwip);
    ether.printIp("DNS: ", ether.dnsip);

    if (!ether.dnsLookup(website))
      printf_P(PSTR("DNS failed"));

    ether.printIp("SRV: ", ether.hisip);

    break;
  }
}
```

i'm running packetLoop in first while `while(millis() - startedAt < 10000)`. The problem is when it goes out of if(RTC.alarm(ALARM_2)) and try to call dnsLookUp. Should I put `ether.packetLoop(ether.packetReceive());` in void loop ? after // end if RTC.alarm

Thank you

solarkennedy, 2017-06-29:
I'm pretty sure in this case you don't need to do a `packetRecieve`.

But you should know that dnsLookup is blocking and will disgard anything other than the dns response
https://github.com/jcw/ethercard/blob/3af09e38129c43597f4e0febd37e9c838253249f/dns.cpp#L91

While I don't think your code is technically wrong, 1 dns lookup every minute gives a lot of chances to fail. dns is udp, if the packet is dropped, then it is done. 

Looking at the implementation of `dnsLookup`, I don't see anything that would stop this from working multiple times during a session. My only advice might be to add more debug in that function to see what exactly is hanging up, or take an alternative approach of resetting the whole microcontroller, watchdog-style, instead of trying to re-initialize the tcp stack.