njh/EtherCard #283: Gratuitous ARP support

thekakester, 2017-08-26:
Ethercard does not support Gratuitous ARP packets.
I'm new to Gratuitous ARP, so I'll try to summarize best I can.

When we need to talk to a device, we need their MAC address.
Most of the time, we get this by sending out a broadcast asking "Hey <IP>, what's your MAC?" and then the device with that IP sends back its MAC.

However, some (feisty) devices don't do it this way.  Instead, they just broadcast "Hey everyone, my MAC is <MAC>, just in case you wanted to know".  This is called Gratuitous ARP, sending the MAC before you're asked for it.

**How I found this bug**
I am using an Allen Bradley MicroLogix 1400 PLC with an ethernet port that hosts a web page.  If I connect the PLC to a router, and my EtherCard device to the same router, everything works fine.

Why?  Because the EtherCard device asks the gateway (the router) for the MAC address, and the router has it (because the router supports Gratuitous ARP).  Long story short, I can scrape the web page, so long as both devices are connected to a router.

When the devices are connected directly to each other, EtherCard never resolves the MAC address, and therefore never communicates.  Meanwhile, the Allen Bradley PLC keeps shouting "Hey! This is my Mac Address", but unfortunately EtherCard doesn't know what to do with this.

**Example of ARP broadcast before it's expected.**
First chunk is "broadcast" (FF FF FF FF FF FF)
Next chunk is my MAC (01 1D 9C A3 5F 4C)
Next chunk is 0806 which means Type: ARP
If you care about the rest, here's the breakdown.  (PS, the IP for this device is 192.168.0.2)
[image]

**HOW TO FIX**
In EtherCard::packetLoop(), check for Gratuitous ARP Broadcasts, and handle them.
This should ONLY be done if we can guarantee there is only one device connected (no router or switch).  It would not be intelligent to store ALL ARP broadcasts because we only really care about one (our destination).  If we're on a switch, we will get broadcasts from every device on the network.

I will try my best to commit a fix to this.  I figured I'd write up a detailed log just in case I never get around to it.

ifreislich, 2017-08-30:
I know this is a kludge, but for now try configuring your PLC IP as the default gateway for your arduino and see if that at least populates the arduino's ARP table.  I'd be hugely surprised if the PLC doesn't respond to an ARP request for its address.  It's common for systems to send gratuitous ARPs on events such as link state changes, acquisition of an IP address, etc.

Memory is short on microcontrollers, but it's not unreasonable to maintain an ARP table of 2 or 3 MACs which get cycled out due to use.

The snippet below confirms that the ethercard library does send ARP requests for an address on the local network, however it only asks once, doesn't wait for a response and then sends all traffic to the router. Ignore the bad checksums, it's due to hardware checksum offload and tcpdump.  I need to debug further in my Copious Free Time.

10.0.19.0/24 network
10.0.19.1 - router/dns
10.0.19.3 - destination host, but this host is not connected so there will never be an ARP response
10.0.19.20 - the arduino

What is my router MAC?
20:59:05.964281 1a:2b:3c:4d:5e:6f > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 64: Ethernet (len 6), IPv4 (len 4), Request who-has 10.0.19.1 tell 10.0.19.20, length 50
20:59:05.964288 d8:eb:97:b3:1f:0d > 1a:2b:3c:4d:5e:6f, ethertype ARP (0x0806), length 42: Ethernet (len 6), IPv4 (len 4), Reply 10.0.19.1 is-at d8:eb:97:b3:1f:0d, length 28

What is the address of the host I must log to?
20:59:05.965139 1a:2b:3c:4d:5e:6f > d8:eb:97:b3:1f:0d, ethertype IPv4 (0x0800), length 68: (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 50)
    10.0.19.20.57345 > 10.0.19.1.53: 5633+ A? test. (22)
20:59:05.965313 d8:eb:97:b3:1f:0d > 1a:2b:3c:4d:5e:6f, ethertype IPv4 (0x0800), length 116: (tos 0x0, ttl 64, id 57743, offset 0, flags [none], proto UDP (17), length 102, bad cksum 0 (->5ee3)!)
    10.0.19.1.53 > 10.0.19.20.57345: 5633* 1/1/0 test. A 10.0.19.3 (74)

First log packet delivered to the router
20:59:06.066777 1a:2b:3c:4d:5e:6f > d8:eb:97:b3:1f:0d, ethertype IPv4 (0x0800), length 108: (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 90)
    10.0.19.20.5002 > 10.0.19.3.5002: UDP, length 62

Router asks ARP who-has 10.0.19.3
20:59:06.066788 d8:eb:97:b3:1f:0d > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 42: Ethernet (len 6), IPv4 (len 4), Request who-has 10.0.19.3 tell 10.0.19.1, length 28

Arduino asks ARP who-has 10.0.19.3
20:59:06.067234 1a:2b:3c:4d:5e:6f > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 64: Ethernet (len 6), IPv4 (len 4), Request who-has 10.0.19.3 tell 10.0.19.20, length 50

Second log  packet delivered to the router
20:59:06.166583 1a:2b:3c:4d:5e:6f > d8:eb:97:b3:1f:0d, ethertype IPv4 (0x0800), length 108: (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 90)
    10.0.19.20.5002 > 10.0.19.3.5002: UDP, length 62

Router asks ARP who-has 10.0.19.3
20:59:06.166593 d8:eb:97:b3:1f:0d > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 42: Ethernet (len 6), IPv4 (len 4), Request who-has 10.0.19.3 tell 10.0.19.1, length 28

Third log packet delivered to the router
20:59:06.267295 1a:2b:3c:4d:5e:6f > d8:eb:97:b3:1f:0d, ethertype IPv4 (0x0800), length 108: (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 90)
    10.0.19.20.5002 > 10.0.19.3.5002: UDP, length 62

thekakester, 2017-08-30:
I played around with this last night and I discovered that I may be wrong.
I tried setting my gateway as the PLC IP, but that still didn't do anything.

I noticed that my ethernet module does not have any lights (specifically the isLinkUp light) when plugged into the PLC, but it DOES have lights when plugged into my computer or a router.
Just to try to eliminate variables, I tried a different PLC (I have many) and none of them seem to get form a link.

So at this point I'm trying to find out why my computer can communicate with the PLC completely fine, but my ethernet module cannot even establish a link, let alone send packets.
If I find a solution, I will make sure it gets documented somewhere, even if this is the wrong issue to post it to.

ifreislich, 2017-08-30:
The ENC28J60 with ethercard only negotiates 10baseT/UTP <half-duplex>.  It's capable of 10baseT/UTP <full-duplex> but that support has not been turned on by by the ethercard library.  The ENC28J60 does not support auto-MIDX and your PLC probably doesn't either.  Try using a cross-over ethernet cable when you connect them together.  Also try turning off autonegotiation on your PC's ethernet and setting it to 10/half and see what happens when you connect to the PLC.

(Check my pull request #285 to see how to turn the "activity" light into a proper activity light).

thekakester, 2017-08-31:
Here's a quick update.  I'm traveling for the next few weeks, so my access to tools/hardware is limited.  I cut and spliced an ethernet cable to make a crossover (although I accidentally connected green-white to brown-white).  After testing my connections I realized my mistake, but I didn't have time to fix it.  [This Image](http://www.peakelec.co.uk/art/4wirecross.gif) gave me hope that only the orange/green wires mattered.  I plugged it into my arduino and the PLC and the isLinkUp() light came on (and isLinkUp() returns true instead of false).  It was still unable to resolve the MAC for the gateway (PLC), and it never received any incoming packets.  When I get back to my hotel this afternoon, I'll try to wire the ethernet cable correctly to see if that does anything, and then I'll start doing some digging to see what's happening.

I didn't get time to touch any code last night, so I couldn't look into the half-duplex vs full-duplex part you mentioned.  I'll try setting my PC to 10/half when I get back.

Thanks for the advice so far!  I don't know if I'd ever have tried the things you suggested.

thekakester, 2017-09-01:
I just played around with auto-negotiate on my PC.  Here's what I found.
From PC to PLC:
*Autonegotiate: **Works**
*100Mbps Full-Duplex: **Works**
*100Mbps Half-Duplex: **Works**
*10Mbps Full-Duplex: **Fails**
*10Mbps Half-Duplex: **Fails**

It turns out the PLC has some settings too!  (For anyone with the same PLC, it's Settings->Advanced->ENET->Port Settings)
It was defaulted to "Force 100M Full-Duplex" so I set this to 10M Half-duplex... but still no luck.

One last try.  Leaving the PLC at 10M Half-duplex, but this time using my (poorly made) cross-over cable EVERYTHING WORKED!

Now that everything works, I figured I'd push the limits to see what else I can get to work.
Changing Settings on the PLC, using a cross-over ethernet cable:
*10M Half-Duplex: **Works**
*10M Full-Duplex: **Works** (I didn't make changes to the ethercard library either!)
*100M Half-Duplex: **Fails**
*100M Full-Duplex:  **Fails**
*Autonegotiate: **Works**

Thanks a million @ifreislich!  You told me to look in places I didn't know existed!
One final question (which isn't "super" important): Do you know if there's a way to configure ethercard to use a standard ethernet patch cable instead of a crossover?  I know you said that it doesn't support auto-MIDX, but I was just curious if theres a way to manually configure it.  Unfortunately I'm using a pre-built module, so I can't just re-route wires from the IC to the ethernet port.

ifreislich, 2017-09-03:
The only way to do MIDX is to use a cross over cable or to change the connection on the PCB to effect the cross over.  The ENC28J60 chip does not implement cross over in hardware.

closed 2018-07-27 by njh