njh/EtherCard #289: Hang on start with a probability ~50%.

Gomezzzzz, 2017-10-21:
I'm using EtherCard.h to display a current temperature from the file temp.php on a mysite.com (no static IP), which parses current temperature from a weather forecast site.
All is working like a charm but nearly in half of the cases on start (power or reboot) it hangs on 

`
Ethernet controller initialized

DHCP configuration done
DNS resolution done
SRV IP:	xxx.xxx.xxx.xxx

`
and it's all.
Here is my sketch
`
#include <EtherCard.h>

#define HTTP_HEADER_OFFSET 0
static byte mymac[] = {0x12,0x34,0x56,0x78,0x90,0xAB};
const char website[] PROGMEM = "www.mysite.com";
byte Ethernet::buffer[700];
static uint32_t timer = 0;
String fromsrv="",c_temp="",str_fromsrv="";
int retry_get = 0;

static void response_callback (byte status, word off, word len) {
  fromsrv = (const char*) Ethernet::buffer + off;
  str_fromsrv=fromsrv.substring(fromsrv.indexOf("<temp>")+6, fromsrv.indexOf("</temp>"));
  if(str_fromsrv.length()>7){
    retry_get = 0;
    c_temp = str_fromsrv;
    Serial.println(c_temp);
  } else {
    retry_get = 1;
    if(c_temp.length()<1)
      c_temp = "---  ---";
    Serial.println("no data, retry in 1 min");
  }
}

void setup () {
  Serial.begin(9600);
  if (!ether.begin(sizeof Ethernet::buffer, mymac, 10))
  {
    Serial.println("Failed to access Ethernet controller");
    while(1);
  }
  else
    Serial.println("Ethernet controller initialized");
  Serial.println();

  if (!ether.dhcpSetup())
  {
    Serial.println("Failed to get configuration from DHCP");
    while(1);
  }
  else
    Serial.println("DHCP configuration done");

  if (!ether.dnsLookup(website))
  {
    Serial.println("DNS failed");
    while(1);
  }
  else 
    Serial.println("DNS resolution done"); 

  ether.printIp("SRV IP:\t", ether.hisip);
  Serial.println();
}

void loop() {
  ether.packetLoop(ether.packetReceive());
  
  if (retry_get > 0) {//if no answer, try in a minute
    timer = millis() + 60000;
    retry_get = 0;
  }
  if (millis() > timer) {
    timer = millis() + 600000;//10 min interval
    ether.browseUrl(PSTR("/temp.php"), "", website, response_callback);

  }
  show(c_temp);
}

void show(String value) {
    // some "show on a display" code
}
`
Any help appretiated.

chaosdesignnz, 2017-10-21:
I had issues with code i was running for one of my projects, A good place to start is to add "Serial.printin()" debug strings after each step in the code to try and narrow down where the code is hanging.

solarkennedy, 2017-10-21:
Are you sure the ethernet module has a solid 3.3v supply?

Gomezzzzz, 2017-10-21:
**chaosdesignnz**, thanks I'll try it like in PHP in some hard cases. 
**solarkennedy**, no, it has old good 5v supply. ;)
https://aliexpress.com/item/Smart-Electronics-ENC28J60-LAN-Ethernet-Network-Board-Module-25MHZ-Crystal-AVR-51-LPC-STM32-3-3V/32821051354.html
It has dropout voltage regulator 5 -> 3.3

closed 2018-07-28 by njh