njh/EtherCard #290: TCP Equivalent of udpListener?

con247, 2017-11-14:
Hi,

The udpListener sample is exactly what I need for my project (just sending text via UDP then processing it in a function). Is there a sample or any instructions on how to do the same with TCP? I have tried the code below to print a TCP packet sent to port 80 with the modified code below, however it prints the contents to the serial monitor multiple times and sends back TCP responses each time. I am using https://packetsender.com/ to send a packet. How can I setup the code so that I can send a packet to a selectable TCP port and just print the contents like the udpListener?

Note: I re-arranged the code below so that "my" code was at the top for easier visability. The setup routine, etc, are in the same order as the sample udpListener.ino. Thanks in advance for any direction.
```c++
void loop(){
  //this must be called for ethercard functions to work.
  checkEthernet();
}
void checkEthernet() {
  int payloadPos = ether.packetLoop(ether.packetReceive());

  if (payloadPos) {
    Serial.print("Payload Pos: ");
    Serial.println(payloadPos);
    char * incomingData = (char *) Ethernet::buffer + payloadPos;
     Serial.println(incomingData);

  }
}

//callback that prints received packets to the serial port
void udpSerialPrint(uint16_t dest_port, uint8_t src_ip[IP_LEN], uint16_t src_port, const char *data, uint16_t len){
  IPAddress src(src_ip[0],src_ip[1],src_ip[2],src_ip[3]);
  
  Serial.print("dest_port: ");
  Serial.println(dest_port);
  Serial.print("src_port: ");
  Serial.println(src_port);
  
  
  Serial.print("src_port: ");
  ether.printIp(src_ip);
  Serial.println("data: ");
  Serial.println(data);
}

// Demonstrates usage of the new udpServer feature.
//You can register the same function to multiple ports, and multiple functions to the same port.
//
// 2013-4-7 Brian Lee <cybexsoft@hotmail.com>

#include <EtherCard.h>
#include <IPAddress.h>

#define STATIC 0  // set to 1 to disable DHCP (adjust myip/gwip values below)

#if STATIC
// ethernet interface ip address
static byte myip[] = { 192,168,0,200 };
// gateway ip address
static byte gwip[] = { 192,168,0,1 };
#endif

// ethernet mac address - must be unique on your network
static byte mymac[] = { 0x70,0x69,0x69,0x2D,0x30,0x31 };

byte Ethernet::buffer[500]; // tcp/ip send and receive buffer

void setup(){
  Serial.begin(57600);
  Serial.println(F("\n[backSoon]"));

  if (ether.begin(sizeof Ethernet::buffer, mymac) == 0)
    Serial.println(F("Failed to access Ethernet controller"));
#if STATIC
  ether.staticSetup(myip, gwip);
#else
  if (!ether.dhcpSetup())
    Serial.println(F("DHCP failed"));
#endif

  ether.printIp("IP:  ", ether.myip);
  ether.printIp("GW:  ", ether.gwip);
  ether.printIp("DNS: ", ether.dnsip);

  //register udpSerialPrint() to port 1337
  ether.udpServerListenOnPort(&udpSerialPrint, 1337);

  //register udpSerialPrint() to port 42.
  ether.udpServerListenOnPort(&udpSerialPrint, 42);
}

```

mtzfactory, 2018-04-14:
Hey! Hi,

I'm very interested in a TCP server as well, have you got any improvement on this?

I'm stuck with TCP Sequence and Acknowledgment Numbers and that kind of stuff...

Best regards,
Ricardo.

con247, 2018-04-14:
I was trying to use this as a PoC for something at work, but I left the company shortly after this posting, so I didn't pursue it any further. I hope you figure out what you need!